Air-duino/Arduino-AirMCU #4: 希望尽快支持platfromIO，Arduino IDE用着太拉跨。

qnxsgwy, 2023-07-06:
Arduino支持已经挺完善了，platfromIO无非就是多写个配置文件的事，希望尽快支持platfromIO。

zgjsntsm, 2023-07-11:
同求，PIO的体验更好。

lily-Robot, 2023-07-11:
Let me try it

santook, 2023-07-11:
试试Visual studio+Visual Micro吧，PIO下载太拉跨

zgjsntsm, 2023-07-11:
> 

太重了，且只支持arduino

chenxuuu, 2023-07-11:
都开源的，你们可以自己适配，毕竟新建issue的作者也说了，就一个配置文件的事

qnxsgwy, 2023-07-12:
> 都开源的，你们可以自己适配，毕竟新建issue的作者也说了，就一个配置文件的事
> […](#)

但是想收录到Pio官方的支持列表，开箱即用。Pio的Platform，Frameworks，Boards一系列配置文件都需要给官方Pull requests，为了长期问题维护，还是你们官方提Pull更合适吧。

chenxuuu, 2023-07-13:
是的，提交当然是官方来搞，但是配置文件大家可以贡献，毕竟开发人员里没有人用platform io

HalfSweet, 2023-07-16:
> > 都开源的，你们可以自己适配，毕竟新建issue的作者也说了，就一个配置文件的事
> > […  ……](#)
> 
> 但是想收录到Pio官方的支持列表，开箱即用。Pio的Platform，Frameworks，Boards一系列配置文件都需要给官方Pull requests，为了长期问题维护，还是你们官方提Pull更合适吧。

PIO的文档太少了，不太好搞，如果有大佬可以做一套完整的描述文件/CI自动上传包的话可以联系我<HalfSweet@HalfSweet.cn>或者直接提交Pr到仓库，我们来进行上传到官方的包管理系统

目前的PIO的仓库独立在<https://github.com/Air-duino/platform-airmcu>下

jyzhkj, 2023-08-02:
同求呀，PIO好用得多了。

给个例子吧, 肯定有帮助 ：
国内仿avr的芯片lgt 的arduino
arduino：https://github.com/dbuezas/lgt8fx
PIO: https://github.com/darkautism/pio-lgt8fx

BKTM-2020, 2023-08-17:
支持官方移植到platfromIO，这个平台比arduinoide 好很多 现在用的人是越来越多

MaShuilong, 2023-08-17:
适配PIO的时候记得拍一下我

TimeKeepper, 2023-09-08:
aaaaa什么时候适配platformio

HalfSweet, 2023-11-08:
根据pio的官方定价策略：https://registry.platformio.org/pricing

想要发布platform以及tools包，至少得需要Pro会员，但是每月的续费高达10美元，这对于一个社区项目来说成本似乎有些高了。

因此这个计划将长期搁置，直到我有足够的小钱钱能付得起会员，或者有其他赞助来源吧QwQ

~~我就说为什么之前我ci的包死活publish不上去，原来是要加钱啊~~