Air-duino/Arduino-AirMCU #5: Wire库不可用

santook, 2023-07-10:
Wire库无法使用，使用逻辑分析仪检测Datasheet对应的引脚PF0、PF1及PA2、PA3均无信号。

chenxuuu, 2023-07-10:
提供一些复现方法、代码、现象

HalfSweet, 2023-07-17:
> 直接调用例程中的master_write即可复现

可以详细描述下连接的什么外设吗

HalfSweet, 2023-07-17:
> > > 直接调用例程中的master_write即可复现
> > 
> > 
> > 可以详细描述下连接的什么外设吗
> 
> 直接连接逻辑分析仪就可以看出来，不需要连接外设。我看了下应该是HAL那边接口的问题。我测试了合宙的SDK和puya的SDK都有这个问题。

使用中断发送，默认无超时时间，可能会卡死总线，可以连接上外设之后尝试
如果您觉得有必要，可以开个新的issue来增加超时检测

songzhishuo, 2023-07-17:
> > > > 直接调用例程中的master_write即可复现
> > > 
> > > 
> > > 可以详细描述下连接的什么外设吗
> > 
> > 
> > 直接连接逻辑分析仪就可以看出来，不需要连接外设。我看了下应该是HAL那边接口的问题。我测试了合宙的SDK和puya的SDK都有这个问题。
> 
> 使用中断发送，默认无超时时间，可能会卡死总线，可以连接上外设之后尝试 如果您觉得有必要，可以开个新的issue来增加超时检测
大佬的意思是说，这是设计缺陷吗？

songzhishuo, 2023-07-19:
> 请更新到最新的版本，我这边AT24Cxx以及ssd1306测试都无问题，可以正常收发，注意8位和7位地址的区分
ok， 我看到你有提一个EEP的库， 我试试。另外我使用keil的工程做iic 通讯也没有成功。很奇怪

HalfSweet, 2023-07-19:
> > 请更新到最新的版本，我这边AT24Cxx以及ssd1306测试都无问题，可以正常收发，注意8位和7位地址的区分
> > ok， 我看到你有提一个EEP的库， 我试试。另外我使用keil的工程做iic 通讯也没有成功。很奇怪

keil这边实测是没啥问题的，你这边是时序不对还是卡在什么地方了

songzhishuo, 2023-07-19:
> 

不好意思我的测试有误，keil下AT24C02 和Arduino 下AT24C02 均可以。非常抱歉。 但是两个板子I2C_TwoBoard_CommunicationMaster_IT和I2C_TwoBoard_CommunicationSlave_IT 通讯不正常，会卡死在第二个while (HAL_I2C_GetState(&I2cHandle) != HAL_I2C_STATE_READY); 中。

HalfSweet, 2023-07-20:
> > 
> 
> 不好意思我的测试有误，keil下AT24C02 和Arduino 下AT24C02 均可以。非常抱歉。 但是两个板子I2C_TwoBoard_CommunicationMaster_IT和I2C_TwoBoard_CommunicationSlave_IT 通讯不正常，会卡死在第二个while (HAL_I2C_GetState(&I2cHandle) != HAL_I2C_STATE_READY); 中。

你这边有条件使用逻辑分析仪进行时序的抓取吗？目前的话我这边似乎无法复现，工作正常

songzhishuo, 2023-07-23:
> > > 
> > 
> > 
> > 不好意思我的测试有误，keil下AT24C02 和Arduino 下AT24C02 均可以。非常抱歉。 但是两个板子I2C_TwoBoard_CommunicationMaster_IT和I2C_TwoBoard_CommunicationSlave_IT 通讯不正常，会卡死在第二个while (HAL_I2C_GetState(&I2cHandle) != HAL_I2C_STATE_READY); 中。
> 
> 你这边有条件使用逻辑分析仪进行时序的抓取吗？目前的话我这边似乎无法复现，工作正常

很奇怪， 我从新使用同样的代码已经无法复现这个问题。最近测试我会用逻辑分析再挂机测试下，如果再次出现我会同步下截图波形

commit ffd1e689 (2023-07-10): Merge pull request #6 from Air-duino/fix/wire

fix: 修复PinName与宏不对应的问题
close: #5
closed 2023-08-02 by chenxuuu