Air-duino/Arduino-AirMCU #12: AirISP command not found in mac

z2wjswzzw, 2023-07-17:
AirISP command not found in mac

use mac ArduinoIDE to flash rom to air001:
user@MacBook-Pro 1.2.2.0 % "/Users/user/Library/Arduino15/packages/AirM2M/tools/AirISP/1.2.2.0/AirISP" --chip "auto" --port "/dev/cu.usbserial-0001" --baud 115200 --before default_reset --after hard_reset write_flash -e -p 0x08000000 "/private/var/folders/z5/7h2gkqpn3x70bm28dp_l5ktw0000gn/T/arduino/sketches/63087252B0F3D75209FD63EB4F210E8B/sketch_jul17a.ino.bin"

zsh: permission denied: /Users/user/Library/Arduino15/packages/AirM2M/tools/AirISP/1.2.2.0/AirISP
user@MacBook-Pro 1.2.2.0 % sudo "/Users/user/Library/Arduino15/packages/AirM2M/tools/AirISP/1.2.2.0/AirISP" --chip "auto" --port "/dev/cu.usbserial-0001" --baud 115200 --before default_reset --after hard_reset write_flash -e -p 0x08000000 "/private/var/folders/z5/7h2gkqpn3x70bm28dp_l5ktw0000gn/T/arduino/sketches/63087252B0F3D75209FD63EB4F210E8B/sketch_jul17a.ino.bin"

then i into sudo：
sudo: /Users/user/Library/Arduino15/packages/AirM2M/tools/AirISP/1.2.2.0/AirISP: command not found
user@MacBook-Pro 1.2.2.0 % sudo: /Users/Library/Arduino15/packages/AirM2M/tools/AirISP/1.2.2.0/AirISP: command not found 

My ArduinoIDE version：2.0.4
My OS version：13.0 (22A380)

HalfSweet, 2023-07-17:
Please try to save the project to any user directory first and then try again.

z2wjswzzw, 2023-07-17:
it's also have the same problem:
MacBook-Pro 1.2.2.0 % sudo "AirISP" --chip "auto" --port "/dev/cu.usbserial-0001" --baud 115200 --before default_reset --after hard_reset write_flash -e -p 0x08000000 "/private/var/folders/z5/7h2gkqpn3x70bm2_l5ktw0000gn/T/arduino/sketches/0F5B328EE29737EFCB54222A4873D071/sketch_jul17a.ino.bin"
sudo: AirISP: command not found

HalfSweet, 2023-07-17:
That's odd, if it's easier for you, can you download the AirISP tool separately to run it? Use in the terminal<https://github.com/Air-duino/AirISP/releases/tag/1.2.2.0>
```
AirISP -v
```
command in the terminal.

z2wjswzzw, 2023-07-17:
I downloaded (AirISP-osx-arm64, AirISP-osx-x64) and tried the following commands in the terminal, but none of them seemed to work. Is my command wrong?

casillas@MacBook-Pro AirISP-osx-x64 % pwd
/Users/casillas/Downloads/AirISP-osx-x64

casillas@MacBook-Pro AirISP-osx-x64 % AirISP
zsh: command not found: AirISP

casillas@MacBook-Pro AirISP-osx-x64 % ./AirISP
zsh: permission denied: ./AirISP

casillas@MacBook-Pro AirISP-osx-x64 % sudo ./AirISP
sudo: ./AirISP: command not found

casillas@MacBook-Pro AirISP-osx-x64 % AirISP -v    
zsh: command not found: AirISP

casillas@MacBook-Pro AirISP-osx-x64 % sudo ./AirISP -v
sudo: ./AirISP: command not found

z2wjswzzw, 2023-07-17:
I tried it in window10 and it works fine. ^ - ^

HalfSweet, 2023-07-17:
> I downloaded (AirISP-osx-arm64, AirISP-osx-x64) and tried the following commands in the terminal, but none of them seemed to work. Is my command wrong?
>
> casillas@MacBook-Pro AirISP-osx-x64 % pwd [...]

Can you try to set the default rwx using the chmod command? I think it might be a permissions issue that is causing this.

z2wjswzzw, 2023-07-18:
after set the default rwx using the chmod command, a new problem arises:

casillas@MacBook-Pro 1.2.2.0 % ls -l
total 25704
-rwxrwxrwx@ 1 casillas  staff  13063691  7 17 18:44 AirISP
-rwxrwxrwx  1 casillas  staff     16416  7 17 18:44 AirISP.pdb
-rwxrwxrwx@ 1 casillas  staff     70464  7 17 18:44 libSystem.IO.Ports.Native.dylib
casillas@MacBook-Pro 1.2.2.0 % sudo "/Users/casillas/Library/Arduino15/packages/AirM2M/tools/AirISP/1.2.2.0/AirISP" --chip "auto" --port "/dev/cu.usbserial-0001" --baud 115200 --before default_reset --after hard_reset write_flash -e -p 0x08000000 "/private/var/folders/z5/7h2gkqpn3x70bm28dp_l5ktw0000gn/T/arduino/sketches/0F5B328EE29737EFCB54222A4873D071/sketch_jul17a.ino.bin"

zsh: killed     sudo  --chip "auto" --port "/dev/cu.usbserial-0001" --baud 115200 --before

closed 2023-07-28 by chenxuuu