Air-duino/Arduino-AirMCU #20: 希望能够获取Air001芯片温度传感器和内部参考电压值

BuaaHZY, 2023-07-21:
## 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.
Air001芯片ADC中2 个内部模拟输入，可获取芯片温度传感器和内部参考电压，不知道目前是否有函数可以读取，如果没有希望后续能够添加。

## 描述您想要的解决方案 / Describe the solution you'd like
A clear and concise description of what you want to happen.
通过某个函数获取内部模拟输入(温度传感、内部参考电压)

## 描述您考虑过的替代方案 / Describe alternatives you've considered
使用analogRead()函数来读取

## 其他备注信息
试过用analogRead(ATEMP);来获取温度传感器，但没有效果

HalfSweet, 2023-08-09:
此方法一直支持，请使用`analogRead(ATEMP)` `analogRead(AVREF)`等函数来获取温度以及内部参考电压，如果您想转换为可以看到的电压值，请参考如下程序
```cpp
#include "airyyxx_ll_adc.h"

/* Analog read resolution */
#define LL_ADC_RESOLUTION LL_ADC_RESOLUTION_12B
#define ADC_RANGE 4096

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(115200);
  while (!Serial)
    ;
  analogReadResolution(12);
}

static int32_t readVref() {
#ifdef __LL_ADC_CALC_VREFANALOG_VOLTAGE
  return (__LL_ADC_CALC_VREFANALOG_VOLTAGE(analogRead(AVREF), LL_ADC_RESOLUTION));
#else
  return (VREFINT * ADC_RANGE / analogRead(AVREF));  // ADC sample to mV
#endif
}

#ifdef ATEMP
static int32_t readTempSensor(int32_t VRef) {
#ifdef __LL_ADC_CALC_TEMPERATURE
  return (__LL_ADC_CALC_TEMPERATURE(VRef, analogRead(ATEMP), LL_ADC_RESOLUTION));
#elif defined(__LL_ADC_CALC_TEMPERATURE_TYP_PARAMS)
  return (__LL_ADC_CALC_TEMPERATURE_TYP_PARAMS(AVG_SLOPE, VTEMP, CALX_TEMP, VRef, analogRead(ATEMP), LL_ADC_RESOLUTION));
#else
  return 0;
#endif
}
#endif

static int32_t readVoltage(int32_t VRef, uint32_t pin) {
  return (__LL_ADC_CALC_DATA_TO_VOLTAGE(VRef, analogRead(pin), LL_ADC_RESOLUTION));
}

// The loop routine runs over and over again forever:
void loop() {
#if defined(ICACHE) && defined(HAL_ICACHE_MODULE_ENABLED) && !defined(HAL_ICACHE_MODULE_DISABLED)
  bool icache_enabled = false;
  if (HAL_ICACHE_IsEnabled() == 1) {
    icache_enabled = true;
    /* Disable instruction cache prior to internal cacheable memory update */
    if (HAL_ICACHE_Disable() != HAL_OK) {
      Error_Handler();
    }
  }
#endif /* ICACHE && HAL_ICACHE_MODULE_ENABLED && !HAL_ICACHE_MODULE_DISABLED */
  // Print out the value read
  int32_t VRef = readVref();
  Serial.printf("VRef(mv)= %i", VRef);
#ifdef ATEMP
  Serial.printf("\tTemp(°C)= %i", readTempSensor(VRef));
#endif
#ifdef AVBAT
  Serial.printf("\tVbat(mv)= %i", readVoltage(VRef, AVBAT));
#endif
  Serial.printf("\tA0(mv)= %i\n", readVoltage(VRef, A0));
  delay(200);
#if defined(ICACHE) && defined(HAL_ICACHE_MODULE_ENABLED) && !defined(HAL_ICACHE_MODULE_DISABLED)
  if (icache_enabled) {
    /* Re-enable instruction cache */
    if (HAL_ICACHE_Enable() != HAL_OK) {
      Error_Handler();
    }
  }
#endif /* ICACHE && HAL_ICACHE_MODULE_ENABLED && !HAL_ICACHE_MODULE_DISABLED */
}
```

closed 2023-08-09 by HalfSweet

yep96, 2023-09-30:
你好，我使用5V供电，用上面的例程打印出VRef是20V左右，和数据手册说的0-VCC表现并不相同，而直接使用analogRead读取按比例也是20V左右，请问这个可以调整范围吗？

yep96, 2023-09-30:
> 此方法一直支持，请使用`analogRead(ATEMP)` `analogRead(AVREF)`等函数来获取温度以及内部参考电压，如果您想转换为可以看到的电压值，请参考如下程序
> 
> ```c++
> #include "airyyxx_ll_adc.h"
> 
> 
> /* Analog read resolution */
> #define LL_ADC_RESOLUTION LL_ADC_RESOLUTION_12B
> #define ADC_RANGE 4096
> 
> // the setup routine runs once when you press reset:
> void setup() {
>   // initialize serial communication at 9600 bits per second:
>   Serial.begin(115200);
>   while (!Serial)
>     ;
>   analogReadResolution(12);
> }
> 
> static int32_t readVref() {
> #ifdef __LL_ADC_CALC_VREFANALOG_VOLTAGE
>   return (__LL_ADC_CALC_VREFANALOG_VOLTAGE(analogRead(AVREF), LL_ADC_RESOLUTION));
> #else
>   return (VREFINT * ADC_RANGE / analogRead(AVREF));  // ADC sample to mV
> #endif
> }
> 
> #ifdef ATEMP
> static int32_t readTempSensor(int32_t VRef) {
> #ifdef __LL_ADC_CALC_TEMPERATURE
>   return (__LL_ADC_CALC_TEMPERATURE(VRef, analogRead(ATEMP), LL_ADC_RESOLUTION));
> #elif defined(__LL_ADC_CALC_TEMPERATURE_TYP_PARAMS)
>   return (__LL_ADC_CALC_TEMPERATURE_TYP_PARAMS(AVG_SLOPE, VTEMP, CALX_TEMP, VRef, analogRead(ATEMP), LL_ADC_RESOLUTION));
> #else
>   return 0;
> #endif
> }
> #endif
> 
> static int32_t readVoltage(int32_t VRef, uint32_t pin) {
>   return (__LL_ADC_CALC_DATA_TO_VOLTAGE(VRef, analogRead(pin), LL_ADC_RESOLUTION));
> }
> 
> // The loop routine runs over and over again forever:
> void loop() {
> #if defined(ICACHE) && defined(HAL_ICACHE_MODULE_ENABLED) && !defined(HAL_ICACHE_MODULE_DISABLED)
>   bool icache_enabled = false;
>   if (HAL_ICACHE_IsEnabled() == 1) {
>     icache_enabled = true;
>     /* Disable instruction cache prior to internal cacheable memory update */
>     if (HAL_ICACHE_Disable() != HAL_OK) {
>       Error_Handler();
>     }
>   }
> #endif /* ICACHE && HAL_ICACHE_MODULE_ENABLED && !HAL_ICACHE_MODULE_DISABLED */
>   // Print out the value read
>   int32_t VRef = readVref();
>   Serial.printf("VRef(mv)= %i", VRef);
> #ifdef ATEMP
>   Serial.printf("\tTemp(°C)= %i", readTempSensor(VRef));
> #endif
> #ifdef AVBAT
>   Serial.printf("\tVbat(mv)= %i", readVoltage(VRef, AVBAT));
> #endif
>   Serial.printf("\tA0(mv)= %i\n", readVoltage(VRef, A0));
>   delay(200);
> #if defined(ICACHE) && defined(HAL_ICACHE_MODULE_ENABLED) && !defined(HAL_ICACHE_MODULE_DISABLED)
>   if (icache_enabled) {
>     /* Re-enable instruction cache */
>     if (HAL_ICACHE_Enable() != HAL_OK) {
>       Error_Handler();
>     }
>   }
> #endif /* ICACHE && HAL_ICACHE_MODULE_ENABLED && !HAL_ICACHE_MODULE_DISABLED */
> }
> ```

你好，我使用5V供电，用上面的例程打印出VRef是20V左右，和数据手册说的0-VCC表现并不相同，而直接使用analogRead读取按比例也是20V左右，请问这个可以调整范围吗？