Air-duino/Arduino-AirMCU #24: Air001开发板用UART烧写时报 prepare writing to address 0x8000000 failed.，

dengzuolin, 2023-07-26:
## 描述一下这个bug / Describe the bug
Arduino 2.1.0，使用UART对Air001开发板进行烧写，擦除Flash后报 prepare writing to address 0x8000000 failed.，程序未能正确烧录。

## 复现步骤 / To Reproduce
按照下面的步骤，可以复现bug / Steps to reproduce the behavior:

复现条件：

合宙9块9 Air001开发板
操作系统Windows10
Arduino Board 版本 Air MCU v0.3.3
使用开源的DAP Link烧写器（非合宙出品），烧写器RX/TX分别解Air001开发板的TX/RX，3.3v/GND接Air001开发板的3.3v/GND

复现过程：

新建工程的最少代码
通过DAPLink，使用UART烧录
点击Arduino Upload按钮前， 按住Air001开发板的BOOT键，然后按一下RST键，保持BOOT键不放开
点击Arduino Upload

## 预期的行为 / Expected behavior
程序应该能正常烧录

## 截图 / Screenshots
无

## 日志 / Logs
c:/users/deng/appdata/local/arduino15/packages/airm2m/tools/xpack-arm-none-eabi-gcc/12.2.1-1.2/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: C:\Users\deng\AppData\Local\Temp\arduino\sketches\A385DEF190C667BE1D05FC234C4F6802/Ari001Test.ino.elf has a LOAD segment with RWX permissions
Sketch uses 9748 bytes (29%) of program storage space. Maximum is 32768 bytes.
Global variables use 1072 bytes (26%) of dynamic memory, leaving 3024 bytes for local variables. Maximum is 4096 bytes.
AirISP v1.2.4.0
串口 COM10
连接中...___...
擦除flash中（请耐心等待）...
_start write data ...
prepare writing to address 0x8000000 failed.
【提示】
若遇到因为AirMCU库活工具造成的BUG，请务必上报到下面的网址，以供开发者知晓该问题，并持续跟踪：
https://github.com/Air-duino/Arduino-AirMCU/issues

## 系统信息 / Desktop (please complete the following information):
 - OS: Windows
 - Version 10
 - Arduino 2.1.0
 - Arduino Board 版本 Air MCU v0.3.3

## 其他备注信息 / Additional context
无

HalfSweet, 2023-07-26:
请尝试降低烧录波特率后重试

closed 2023-07-28 by chenxuuu

dengzuolin, 2023-07-28:
> 
波特率降低至9600后问题依旧

chenxuuu, 2023-07-28:
抓取一下串口两根线上的数据

tcwj2008, 2023-08-08:
我换成CP2102的串口工具就可以烧录了

chenxuuu, 2023-09-22:
可以尝试先按住boot，再连线和上电，直到烧录完成再松开按键
如果还不行，就降低波特率

 
是不是只要是串口下载工具就可以使用