Air-duino/Arduino-AirMCU #28: 硬件SPI/I2C无法驱动LCD/OLED屏幕

sunnyguhz, 2023-08-01:
## 描述一下这个bug / Describe the bug
使用U8G2库时，使用硬件SPI或者I2C驱动LCD/OLED屏幕时完全无法使用。软件SPI/I2C正常。
屏幕试过好几个型号/驱动芯片（驱动芯片分别为ST7567,UC1609,SSD1327），确认和LCD/OLED的驱动芯片无关。

## 复现步骤 / To Reproduce
只要用U8G2库并且使用硬件SPI或者硬件I2C驱动LCD/OLED都能复现此问题

## 预期的行为 / Expected behavior
修复此bug

## 截图 / Screenshots
无

## 系统信息 / Desktop (please complete the following information):
版本0.3.3依然存在此问题

LuaRobot, 2023-08-01:
@sunnyguhz: 你好！ :wave:

因为你没有按模板新建issue，所以此issue已被自动关闭！请重新按issue模板新建issue

HalfSweet, 2023-08-02:
我这边实测软硬件驱动ssd1306的oled均可以，如果可以请提供详细的源码以及日志

HalfSweet, 2023-08-04:
@sunnyguhz 如果您那边无法提供复现方法，那么我们将关闭这个issue

sunnyguhz, 2023-08-04:
https://github.com/Air-duino/Arduino-AirMCU/issues/28#issuecomment-1662269159

> 我这边实测软硬件驱动ssd1306的oled均可以，如果可以请提供详细的源码以及日志
https://github.com/Air-duino/Arduino-AirMCU/issues/28#issuecomment-1662269159

sunnyguhz, 2023-08-04:
貌似B站上也有个人也遇到硬件I2C无法点屏的问题，在评论第一个
https://www.bilibili.com/video/BV15V4y1h7Jo/?spm_id_from=333.337.search-card.all.click&vd_source=c57978b7c96179b0ac17acfde8415813

chenxuuu, 2023-08-04:
你新建个新的issue吧

 

 https://www.bilibili.com/video/BV15V4y1h7Jo/?spm_id_from=333.337.search-card.all.click&amp;vd_source=c57978b7c96179b0ac17acfde8415813

HalfSweet, 2023-08-05:
> 貌似B站上也有个人也遇到硬件I2C无法点屏的问题，在评论第一个 https://www.bilibili.com/video/BV15V4y1h7Jo/?spm_id_from=333.337.search-card.all.click&vd_source=c57978b7c96179b0ac17acfde8415813

此问题已在 https://github.com/Air-duino/Arduino-AirMCU/pull/10 解决

closed 2023-08-07 by chenxuuu