Air-duino/Arduino-AirMCU #32: String.toFloat() 编译报错

cxeo, 2023-08-03:
### 描述一下这个bug / Describe the bug

对字符串 string使用 string.toFloat()会编译报错
而string.toInt()则正常

### 复现步骤 / To Reproduce

用 String 定义一个浮点数，然后使用.toFloat()进行转换

String a="10.234";
Serial.println(a.toFloat());

只用这两行代码就会编译报错。

### 预期的行为 / Expected behavior

修复

### 截图 / Screenshots

无

### 日志 / Logs

无

### 系统 / System

win10

### 软件版本 / Version

Arduino IDE 2.1.0

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.

HalfSweet, 2023-08-03:
arm-none-eabi-gcc的libc采用的是newlib，因此浮点数相关的操作对于Flash的占用可能会略大

后期我们可能会考虑修改Arduino String以减小对资源的占用

chenxuuu, 2023-08-03:
st的包也编译不过去吧，感觉不应该算bug，锅应该给编译器

cxeo, 2023-08-04:
用string flash占用可以用异常大来形容，没多少代码 flash就快满了

cxeo, 2023-08-04:
Air001开发板，加一句 printf("%d,%d,%d\n",xxx）,Flash就会多占用13%

chenxuuu, 2023-08-04:
参考https://github.com/stm32duino/Arduino_Core_STM32/pull/2036 ，只能重写标准库

cxeo, 2023-08-04:
.toFloat()进行转换 编译报错的原因是flash空间用满不够导致的，调用.toFloat()会占用大量Flash。

HalfSweet, 2023-08-04:
> 参考[stm32duino/Arduino_Core_STM32#2036](https://github.com/stm32duino/Arduino_Core_STM32/pull/2036) ，只能重写标准库

如果是newlib造成的问题，我们恐怕得重新实现一份字符串格式化的库

也许<mtfmt.cc>是个不错的解决方案

closed 2023-08-09 by chenxuuu