Air-duino/Arduino-AirMCU #34: i2c Wire库slave模式异常

xizeyoupan, 2023-08-04:
### 描述一下这个bug / Describe the bug

两份代码，一份主机，一份从机。由于手头只有一块air001开发板，故另一板子使用esp32开发板代替。
esp32 sad:21 scl:22 
air001 sda:pf0 scl:pf1 均为默认。
sad 和 scl 分别通过3k上拉到3V3。

### 复现步骤 / To Reproduce

主机代码：
```
#include <Arduino.h>
#include <Wire.h>

void setup() {
    Serial.begin(115200);

    Wire.setClock(100000);
    Wire.begin();
}

void loop() {

    for (uint8_t i = 0; i < 128; i++) {
        Wire.beginTransmission(8);
        Wire.write(i);
        int code = Wire.endTransmission();
        Serial.println(i);
        Serial.println(code);
        delay(5);
    }
}
```

从机代码：
```
#include <Arduino.h>
#include <Wire.h>

#define ADDRESS (0x08)

void receiveEvent(int numBytes) {
    uint8_t cmd;
    while (Wire.available()) {
        cmd = Wire.read();
        Serial.println(cmd);
    }
}

void setup() {
    Serial.begin(115200);

    Wire.setClock(100000);
    Wire.begin(ADDRESS);
    Wire.onReceive(receiveEvent);
}

void loop() {
    delay(1);
}
```

### 如果正常，应该是什么样 / Expected behavior

在使用air001为主机，esp32为从机时，输出正常：
[image]
[image]

但是使用air001为从机，esp32为主机时，输出每隔一个请求就会丢失：
[image]
[image]

多次尝试依旧如此。

### 截图 / Screenshots

上方已给出。

### 日志 / Logs

没有

### 系统 / System

Win11

### PACK包版本 / Version

0.4.0, Arduino IDE 2.1.0

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

cxeo, 2023-08-05:
同样碰到此BUG，Air001作为从机，通信速度高于40K就容易丢数据，只能降低到20K使用，而Air001手册的I2C是支持400K的。还有一个问题，Air001作为从机，有一定概率I2C卡死，而且主机那里还能收到Air001的ACK信号，但AIr001的onReceive函数就是不被触发，只有复位才能解决。而使用ESP32的I2C库从来都不会这样。

HalfSweet, 2023-08-17:
您这边air001的主频配置是多高呢

xizeyoupan, 2023-08-18:
原先使用是默认的频率4M，经过测试发现主频提高到HSI 16M，CLK 16M后上面的代码正常了。但运行下面的代码依然会卡住：
air001 从机：
```c
#include <Arduino.h>
#include <Wire.h>

#define ADDRESS (0x08)

void receiveEvent(int numBytes) {

}

void requestEvent() {
    Wire.write(0);
}

void setup() {

    Wire.begin(ADDRESS);
    Wire.onReceive(receiveEvent);
    Wire.onRequest(requestEvent);

}

void loop() {

}
```

esp32主机：
```c
#include <Arduino.h>
#include <Wire.h>

void setup() {
    Serial.begin(115200);

    Wire.setClock(100000);
    Wire.begin();
}

void loop() {

    Wire.requestFrom(8, 1);
    while (Wire.available()) {
        int c = Wire.read();
        Serial.print(c);
    }

    delay(100);
}
```

HSI 提高到24M也一样。手头没有外部晶振和逻辑分析仪，其他的懒得测了QAQ

HalfSweet, 2023-08-18:
可能依然是主频不够导致嵌套中断最终卡死，能否尝试修改为HSI 24M PCLK 48M的配置呢

xizeyoupan, 2023-08-18:
试过了，依旧卡死

closed 2023-08-19 by HalfSweet

HalfSweet, 2023-08-19:
补充一份测试代码

Air001

```cpp
#include <Arduino.h>
#include <Wire.h>

#define ADDRESS (0x08)

void receiveEvent(int numBytes) {
  // 循环读取数据(除了最后一个字符)
  while (1 < Wire.available()) {
    // 接收字节数据并赋值给变量c(char)
    char c = Wire.read();
    // 打印该字节
    Serial.print(c);
  }
  // 以int整数的形式接受字节数据并赋值给x(int)
  char x = Wire.read();
  // 打印该int变量x
  Serial.println(x);
}

void requestEvent() {
  Wire.write('1');
  Wire.write('1');
  Wire.write('4');
}

void setup() {
  Serial.begin(115200);
  Wire.begin(ADDRESS);
  Wire.onReceive(receiveEvent);
  Wire.onRequest(requestEvent);
}

void loop() {
}
```

ESP32

```cpp
#include <Arduino.h>
#include <Wire.h>

void setup() {
  Serial.begin(115200);

  Wire.setClock(100000);
  Wire.begin();
}

void loop() {
  Wire.beginTransmission(0x08);
  Wire.write("1919810");
  Serial.println(Wire.endTransmission());
  Wire.requestFrom(8, 3);
  Serial.printf("Wire.available is %d \n", Wire.available());
  Serial.print("read data is ");
  while (Wire.available() > 0) {
    char c = Wire.read();
    Serial.print(c);
    Serial.print(' ');
  }
  Serial.print('\n');
  delay(500);
}
```