Air-duino/Arduino-AirMCU #37: Air001中断冲突

Deerczl, 2023-08-08:
### 描述一下这个bug / Describe the bug

开发平台 Arduino
开发板管理器 AirMCU v0.4.0

目的：通过按键中断亮灯
描述：PA6, PB6分别接两个按键，按下为高。PF1接LED，PF1为高时LED亮。

### 复现步骤 / To Reproduce

最小化工程代码
```
void setup() {
  // put your setup code here, to run once:
  pinMode(PB6, INPUT);
  pinMode(PA6, INPUT);
  pinMode(PF1, OUTPUT);
  attachInterrupt(PA6, button0_ring, RISING);
  attachInterrupt(PB6, button1_ring, RISING);
}

void loop() {

}

void button0_ring() {
  int duty = 40;
  while (digitalRead(PA6) == HIGH) {
    digitalWrite(PF1, HIGH);
    delayMicroseconds(duty);
    digitalWrite(PF1, LOW);
    delayMicroseconds(duty);
  }
}  

void button1_ring() {
  int duty = 40;
  while (digitalRead(PB6) == HIGH) {
    digitalWrite(PF1, HIGH);
    delayMicroseconds(duty);
    digitalWrite(PF1, LOW);
    delayMicroseconds(duty);
  }
}  
```

### 如果正常，应该是什么样 / Expected behavior

下载后仅PB6可以响应中断，PA6无法进入中断。

交换以下语句顺序，
``` 
attachInterrupt(PA6, button0_ring, RISING);
attachInterrupt(PB6, button1_ring, RISING);
```
为
```
attachInterrupt(PB6, button1_ring, RISING);
attachInterrupt(PA6, button0_ring, RISING);
```

下载后仅PA6可以响应终端，PB6无法进入终端

### 截图 / Screenshots

NA

### 日志 / Logs

NA

有任何需要补充的信息请与我联系。

### 系统 / System

Arduino AirMCU

### PACK包版本 / Version

v0.4.0

### 验证

- [X] 检查该问题是否已被提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

chenxuuu, 2023-08-08:
`PA6`与`PB6`共用`EXTI6`，应该是库把`EXTI6`当一个用了

> 手册图 12-2 外部中断/事件 GPIO 映像
[image]

Deerczl, 2023-08-09:
感谢回复！
[image]
按照wiki中的描述，我可否使用PF4规避该EXTI冲突呢。
根据文档描述，PF4仅作为BOOT0使用，但请问其是否还具有输入/输出/中断等功能呢？
[image]

Deerczl, 2023-08-09:
pinName中似乎也没有定义PF4，若无法作为IO使用，建议在wiki中删除PF4标识。

HalfSweet, 2023-08-09:
当然，您可以使用PF4作为中断
pinName确实没有定义，我们将在随后的版本中修复

HalfSweet, 2023-08-09:
作为解决方案，您可以尝试使用`PF_4`来进行定义

Deerczl, 2023-08-10:
感谢答复！

我尝试了下面三种方法，但下载后均无法调用PF_4中断。可否帮忙核实一下代码是否有误，谢谢~

方法一：
```
pinMode(PF_4, INPUT);
attachInterrupt(PF_4, button1_ring, RISING);
```

方法二：
```
air_interrupt_enable(GPIOF, GPIO_PIN_4, button1_ring, GPIO_MODE_IT_RISING);
```

方法三：
```
void APP_ConfigureEXTI() {
  GPIO_InitTypeDef GPIO_InitStruct;
  __HAL_RCC_GPIOF_CLK_ENABLE();                 /* 使能GPIOF时钟 */
  GPIO_InitStruct.Mode = GPIO_MODE_IT_RISING;  /* GPIO模式为上升沿中断 */
  GPIO_InitStruct.Pull = GPIO_PULLDOWN;         /* 下拉 */
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH; /* 速度为高速 */
  GPIO_InitStruct.Pin = GPIO_PIN_4;
  HAL_GPIO_Init(GPIOF, &GPIO_InitStruct);

  /* 使能EXTI中断 */
  HAL_NVIC_EnableIRQ(EXTI4_15_IRQn);
  /* 配置中断优先级 */
  HAL_NVIC_SetPriority(EXTI4_15_IRQn, 0, 0);
}

void EXTI4_15_IRQHandler()
{
  HAL_GPIO_TogglePin(GPIOF, GPIO_PIN_1);
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_4);
}
```

chenxuuu, 2023-08-10:
```cpp
attachInterrupt(digitalPinToInterrupt(PF_4), onChange, CHANGE);
```

closed 2023-08-10 by HalfSweet