Air-duino/Arduino-AirMCU #38: 希望官方能添加一个arduino的LPTIM example

tcwj2008, 2023-08-08:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

因为想快速开发一些功能选择了Air001+Arduino，但是试了一天LPTIM都没成功，对MDK不太熟悉啊。HAL的示例里面又是很多对于时钟的设定，但是Arduino貌似是在Tools里设定的，这里就有点不知道该咋整了。
所以希望官方能封装个LPTIM，或者给个示例，感恩

### 描述您想要的解决方案 / Describe the solution you'd like

所以希望官方能封装个LPTIM，或者给个示例，感恩

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

可以直接在File里调用example

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

BKL5, 2023-10-09:
我试了下HAL的RTC库在arduino里面只有部分可以用。没定时器arduino根本没法玩。

HalfSweet, 2023-10-09:
> 我试了下HAL的RTC库在arduino里面只有部分可以用。没定时器arduino根本没法玩。

部分可用指的是？

profacetom, 2023-12-13:
看了手册air001有9个定时器，TM3 TM4都是通用定时器，希望尽快出arduino的TIMER库，还要RTC库，太常用的功能了，针对air001的硬件驱动库太少了。

HalfSweet, 2023-12-15:
> 看了手册air001有9个定时器，TM3 TM4都是通用定时器，希望尽快出arduino的TIMER库，还要RTC库，太常用的功能了，针对air001的硬件驱动库太少了。

参考<https://arduino.luatos.com/library/timer.html>

closed 2023-12-15 by HalfSweet