Air-duino/Arduino-AirMCU #47: air001低功耗模式

rookieslans, 2023-08-14:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

希望能出一个用arduino写的air001的超低功耗模式的例程。

### 描述您想要的解决方案 / Describe the solution you'd like

希望能出一个用arduino写的air001的超低功耗模式的例程。

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

希望能出一个用arduino写的air001的超低功耗模式的例程，例如一个声控灯：平常处于低功耗模式，检测到有声音时退出低功耗，LED亮一段时间然后再进入低功耗模式。
我不会用arduino配置低功耗，希望可以出个例程让我学习一下，感谢

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

60999, 2023-08-17:
手册中也没写低功耗模式耗电数据。

MethodJiao, 2023-08-29:
低功耗模式对我很重要，目前资料中缺少arduino IDE的开发例程，建议添加

lingchen-jiager, 2023-09-15:
需要arduino编程低功耗模式+1

SendToSouthEast, 2023-09-15:
> 手册中也没写低功耗模式耗电数据。

有的，在datasheet里面有，有典型电流

MethodJiao, 2023-09-26:
这个还没解决么。蹲了两个月

chenxuuu, 2023-09-26:
没人设计接口定义、使用流程

lable, 2023-11-02:
需要这个功能+1

chenxuuu, 2023-11-03:
有低功耗需求的应该用keil的sdk，很难想象001使用arduino做项目，空间都不够

lable, 2023-11-28:
> 有低功耗需求的应该用keil的sdk，很难想象001使用arduino做**量产项目**，空间都不够

我这里儿就有这需求，虽然量不大

sunnyguhz, 2023-12-03:
同需要低功耗模式，同时需要更低速的运行模式，比如1M、2M等

SendToSouthEast, 2023-12-10:
> 同需要低功耗模式，同时需要更低速的运行模式，比如1M、2M等

应该跑不了这么低，内部晶振，我看最低是跑4mhz，你用HAL库函数将HSI配置为4m，不开PLL就行了
arduino中也能直接用hal库函数

lable, 2023-12-14:
尝试用LL库实现了LPTIM定时唤醒低功耗，可以测试下看看能不能用。

[air001-stop.zip](https://github.com/Air-duino/Arduino-AirMCU/files/13667234/air001-stop.zip)

caoxuan19961206, 2023-12-14:
需要这个功能+1

sunnyguhz, 2024-02-11:
> > 同需要低功耗模式，同时需要更低速的运行模式，比如1M、2M等
> 
> 应该跑不了这么低，内部晶振，我看最低是跑4mhz，你用HAL库函数将HSI配置为4m，不开PLL就行了 arduino中也能直接用hal库函数

不知道Air001的系统时钟部分有没有分频器，STM32G0系列我是通过修改分频器配置达到更低的运行速度（1M，2M等等）的。尝试自己修改了下Air001的时钟配置，但是貌似不起作用。

xky183, 2024-04-08:
关注低功耗API的进展，也需要

caoxuan19961206, 2024-04-08:
蹲更

zengcym, 2024-10-25:
很奇怪，我用arduino 让air001进入stop模式，电流还有200uA，是什么情况，难道是我的姿势不对？

why9423, 2025-04-10:
我直接在arduino使用
HAL_PWR_EnterSLEEPMode(PWR_SLEEPENTRY_WFI); 
成了