Air-duino/Arduino-AirMCU #48: 使用外部24M晶振有问题

jyzhkj, 2023-08-15:
### 描述一下这个bug / Describe the bug

air001  
选择外部晶振，外部晶振是24M时，下载程序后，不起振。
不管是PLL倍频。

实测发现：

  RCC_OscInitStruct.HSEFreq = RCC_HSE_16_32MHz;   
这一句有问题。
改为RCC_HSE_8_16MHz 能跑，当然频率是不对的。

### 复现步骤 / To Reproduce

在菜单中选择  
HSE 24Mhz, HCLK 24Mhz
或
HSE 24Mhz, HCLK 48Mhz

### 如果正常，应该是什么样 / Expected behavior

x

### 截图 / Screenshots

x

### 日志 / Logs

x

### 系统 / System

win10 

### PACK包版本 / Version

0.4.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

jyzhkj, 2023-08-15:
为什么关闭问题？
是我搞错了？你没有复现吗？

chenxuuu, 2023-08-15:
空那么多项目我还以为是捣乱的

HalfSweet, 2023-08-17:
RCC_HSE_8_16MHz和RCC_HSE_16_32MHz仅会影响晶振的驱动能力，如果方便的话，是否可以提供一下您的原理图和pcb布线，以及晶体的手册？

根据我的实测，24M的晶体是可以正常工作的

HalfSweet, 2023-08-17:
使用HSE24M PCLK24M和HSE24M PCLK24M都无法进行启动吗？您使用的是哪一种型号的晶振呢

jyzhkj, 2023-08-17:
HSE 24Mhz, HCLK 24Mhz
HSE 24Mhz, HCLK 48Mhz
32Mhz 未测
这3个选项都不行。
晶振就是普通的49S,24M
[image]

**注意同一块板在keil中很正常**
跟踪到 variants\air001\air001_dev中的generic_clock.c文件
SystemClock_Config()函数的内容，对比keil基本一致

另外写了个小函数在arduino启动完之后再setup中调用：
`void APP_SystemClockConfig3(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
  /* 振荡器配置 */
  RCC_OscInitStruct.OscillatorType |= RCC_OSCILLATORTYPE_HSE; /* 选择RCC振荡器为HSE */
  RCC_OscInitStruct.HSIState = RCC_HSI_OFF;
  /* RCC_OscInitStruct.HSIDiv = RCC_HSI_DIV4; */ /* 4分频 */
  // RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_24MHz;                       /* 配置HSI输出时钟为8MHz */
  RCC_OscInitStruct.HSEState = RCC_HSE_ON; /* 开启HSE */
  // RCC_OscInitStruct.HSEFreq = RCC_HSE_8_16MHz;          /* HSE晶振工作频率8M~16M */
  RCC_OscInitStruct.HSEFreq = RCC_HSE_16_32MHz;         /* HSE晶振工作频率16M~32M */
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_OFF;         /* 关闭PLL */
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_NONE; /* 不选择PLL源 */
  /* 配置振荡器 */
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  /* 时钟源配置 */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_PCLK1; /* 选择配置时钟 HCLK,SYSCLK,PCLK1 */
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSE;                                         /* 选择HSE作为系统时钟 */
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;                                             /* AHB时钟 1分频 */
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;                                              /* APB时钟 1分频 */
  /* 配置时钟源 */
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
}`

发现只要是设成"RCC_HSE_16_32MHz" ,就会死。
所以是不是驱动底层的问题？

HalfSweet, 2023-08-18:
尝试注释掉所有关于HSI的配置，看看是否可以正常工作

jyzhkj, 2023-08-18:
没作用，还是死.
相同代码在keil中正常：

 `void APP_SystemClockConfig3(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};
  /* 振荡器配置 */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE; /* 选择RCC振荡器为HSE */
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;                   /* 开启HSE */
  RCC_OscInitStruct.HSEFreq = RCC_HSE_16_32MHz;              /* HSE晶振工作频率16M~32M */
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_OFF;              /* 关闭PLL */
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_NONE;      /* 不选择PLL源 */
  /* 配置振荡器 */
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  /* 时钟源配置 */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_PCLK1; /* 选择配置时钟 HCLK,SYSCLK,PCLK1 */
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSE;                                         /* 选择HSE作为系统时钟 */
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;                                             /* AHB时钟 1分频 */
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;                                              /* APB时钟 1分频 */
  /* 配置时钟源 */
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1) != HAL_OK)
  {
    Error_Handler();
  }
}`

jyzhkj, 2023-08-18:
HAL_RCC_OscConfig  函数 返回值=3   timeout

chenxuuu, 2023-08-22:
最简单的办法，邮寄寄一套测试硬件给开发者

jyzhkj, 2023-09-06:
直接来个boards.txt中的配置吧：
解决的问题：
1:外部晶振问题，其实是 generic_clock.c中 #undef  不起作用引起的。选择外部晶振后，程序中打印一下HSE_VALUE就明白了，全是8M
2:增加JLINK支持
3：mcu flash/ram  可选

menu.cpu=Processor

###############################
# Py32All Dev
###############################

Py32All.name=Py32all

Py32All.menu.cpu.64k=64k/8k
Py32All.menu.cpu.64k.upload.maximum_size=65536
Py32All.menu.cpu.64k.upload.maximum_data_size=8192

Py32All.menu.cpu.32k=32k/4k
Py32All.menu.cpu.32k.upload.maximum_size=32768
Py32All.menu.cpu.32k.upload.maximum_data_size=4096

Py32All.menu.upload_method.AirMethod=AirISP
Py32All.menu.upload_method.AirMethod.upload.tool=AirISP

Py32All.menu.upload_method.jLinkMethod=JLINK
Py32All.menu.upload_method.jLinkMethod.upload.protocol=jlink
Py32All.menu.upload_method.jLinkMethod.upload.tool=jlink_upload
Py32All.menu.upload_method.jLinkMethod.build.upload_flags=-DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG
Py32All.menu.upload_method.jLinkMethod.upload.device=CORTEX-M0+   

Py32All.build.core=AirMCU
Py32All.build.board=AIR001_DEV
Py32All.build.mcu=cortex-m0plus
Py32All.build.series=AIR001xx
Py32All.build.cmsis_lib_gcc=arm_cortexM0l_math
Py32All.build.product_line=AIR001
Py32All.build.variant=AIR001/AIR001_DEV
Py32All.build.air_extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial}
Py32All.build.drivers_include="-I{build.system.path}/Arduino-Air001-Drivers/{build.series}_HAL_Driver/Inc" "-I{build.system.path}/Arduino-Air001-Drivers/{build.series}_HAL_Driver/Src" "-I{build.system.path}/{build.series}" "-I{build.system.path}/Arduino-Air001-Drivers/CMSIS/Device/{build.series}/Include/" "-I{build.system.path}/Arduino-Air001-Drivers/CMSIS/Device/{build.series}/Source/gcc/" 

Py32All.menu.UploadSpeed.115200=115200
Py32All.menu.UploadSpeed.115200.upload.speed=115200
Py32All.menu.UploadSpeed.57600=57600
Py32All.menu.UploadSpeed.57600.upload.speed=57600
Py32All.menu.UploadSpeed.38400=38400
Py32All.menu.UploadSpeed.38400.upload.speed=38400
Py32All.menu.UploadSpeed.19200=19200
Py32All.menu.UploadSpeed.19200.upload.speed=19200
Py32All.menu.UploadSpeed.9600=9600
Py32All.menu.UploadSpeed.9600.upload.speed=9600
Py32All.menu.UploadSpeed.4800=4800
Py32All.menu.UploadSpeed.4800.upload.speed=4800
Py32All.menu.UploadSpeed.2400=2400
Py32All.menu.UploadSpeed.2400.upload.speed=2400
Py32All.menu.UploadSpeed.1200=1200
Py32All.menu.UploadSpeed.1200.upload.speed=1200

# HSI/HSE选择和主频
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M=HSI 4Mhz, HCLK 4Mhz
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M.build.temp=-D{build.series}_HSI_4M_HCLK_4M
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M.build.f_cpu=4000000L
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M.build.HSI_VALUE=4000000L
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M.build.HSE_VALUE=4000000L

Py32All.menu.ClockSourceAndFrequency.HSI8M_HCLK8M=HSI 8Mhz, HCLK 8Mhz
Py32All.menu.ClockSourceAndFrequency.HSI8M_HCLK8M.build.temp=-D{build.series}_HSI_8M_HCLK_8M
Py32All.menu.ClockSourceAndFrequency.HSI8M_HCLK8M.build.f_cpu=8000000L
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M.build.HSI_VALUE=8000000L
Py32All.menu.ClockSourceAndFrequency.HSI4M_HCLK4M.build.HSE_VALUE=8000000L
[... 97 lines cut ...]
Py32All.menu.opt.ogstd.build.flags.optimize=-Og
Py32All.menu.opt.o0std=No Optimization (-O0)
Py32All.menu.opt.o0std.build.flags.optimize=-O0

Py32All.menu.SupplyVoltage.3V3=3.3V
Py32All.menu.SupplyVoltage.3V3.build.sv_tmp=-DVDD_3V3
Py32All.menu.SupplyVoltage.1V8=1.8V
Py32All.menu.SupplyVoltage.3V3.build.sv_tmp=-DVDD_1V8
Py32All.menu.SupplyVoltage.2V5=2.5V
Py32All.menu.SupplyVoltage.3V3.build.sv_tmp=-DVDD_2V5
Py32All.menu.SupplyVoltage.2V8=2.8V
Py32All.menu.SupplyVoltage.3V3.build.sv_tmp=-DVDD_2V8
Py32All.menu.SupplyVoltage.3V8=3.8V
Py32All.menu.SupplyVoltage.3V3.build.sv_tmp=-DVDD_3V8
Py32All.menu.SupplyVoltage.5V=5V
Py32All.menu.SupplyVoltage.5V.build.sv_tmp=-DVDD_5V

# boot
Py32All.menu.BootConnection.defaule=default isp circuit
Py32All.menu.BootConnection.defaule.upload.boot=default_reset
Py32All.menu.BootConnection.direct=direct connect
Py32All.menu.BootConnection.direct.upload.boot=direct_connect

Py32All.build.extra_flags={build.temp} -DF_CPU={build.f_cpu} -DHSI_VALUE={build.HSI_VALUE} -DHSE_VALUE={build.HSE_VALUE} {build.lsc_temp} {build.sv_tmp} -DAIR001_DEV
# Upload menu

HalfSweet, 2023-09-06:
好的，问题已收到，我们会尽快解决

py32我个人后期会单独开一个仓库进行维护，此组织仅作为AirMCU的交流

closed 2023-11-08 by HalfSweet