Air-duino/Arduino-AirMCU #49: arduino软串口有问题

60999, 2023-08-16:
### 描述一下这个bug / Describe the bug

用的官方开发板库自带例子。TwoPortReceive.ino
比如其中，SoftwareSerial portOne(10, 11);，其中10，11是错的吧？
如果改成SoftwareSerial portOne(PB2, PA7);，也无法通过。

### 复现步骤 / To Reproduce

```
完整代码：
/*
  Software serial multiple serial test

 Receives from the two software serial ports,
 sends to the hardware serial port.

 In order to listen on a software port, you call port.listen().
 When using two software serial ports, you have to switch ports
 by listen()ing on each one in turn. Pick a logical time to switch
 ports, like the end of an expected transmission, or when the
 buffer is empty. This example switches ports when there is nothing
 more to read from a port

 The circuit:
 Two devices which communicate serially are needed.
 * First serial device's TX attached to digital pin 10(RX), RX to pin 11(TX)
 * Second serial device's TX attached to digital pin 8(RX), RX to pin 9(TX)

 created 18 Apr. 2011
 modified 19 March 2016
 by Tom Igoe
 based on Mikal Hart's twoPortRXExample

 This example code is in the public domain.

 */

#include <SoftwareSerial.h>
// software serial #1: RX = digital pin 10, TX = digital pin 11
SoftwareSerial portOne(PB2, PA7);

// software serial #2: RX = digital pin 8, TX = digital pin 9
// on the Mega, use other pins instead, since 8 and 9 don't work on the Mega
SoftwareSerial portTwo(PA13, PA14);

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // Start each software serial port
  Serial.println("port one begin:");
  portOne.begin(115200);
  Serial.println("port two begin:");
  portTwo.begin(115200);
}

void loop() {
  // By default, the last initialized port is listening.
  // when you want to listen on a port, explicitly select it:
  portOne.listen();
  Serial.println("Data from port one:");
  // while there is data coming in, read it
  // and send to the hardware serial port:
  while (portOne.available() > 0) {
    char inByte = portOne.read();
    Serial.write(inByte);
  }

  // blank line to separate data from the two ports:
  Serial.println();

  // Now listen on the second port
  portTwo.listen();
  // while there is data coming in, read it
  // and send to the hardware serial port:
  Serial.println("Data from port two:");
  while (portTwo.available() > 0) {
    char inByte = portTwo.read();
    Serial.write(inByte);
  }

  // blank line to separate data from the two ports:
  Serial.println();
}

```

### 如果正常，应该是什么样 / Expected behavior

→ port one begin:
 → port two begin:

### 截图 / Screenshots

[2023-08-16 15:56:07.510] → port one begin:

### 日志 / Logs

[2023-08-16 15:56:07.510] → port one begin:

### 系统 / System

win10

### PACK包版本 / Version

0.4.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

chenxuuu, 2023-08-17:
用9600或更低的波特率试试

HalfSweet, 2023-08-18:
我们测试中暂时没有进行复现，它工作得很好，也许您需要提供更详细的日志和截图

closed 2023-08-19 by HalfSweet