Air-duino/Arduino-AirMCU #50: BUG: 串口接收数据必定丢包。

MaShuilong, 2023-08-17:
### 描述一下这个bug / Describe the bug

使用ArduinoIDE2.1.1环境下 串口波特率112500  开发板库版本号目前最新AirMCU(0.4.1)，望尽快修复。

### 复现步骤 / To Reproduce

//烧录该程序 然后打开串口助手向单片机发送数据 单片机会返回接收到的数据，会丢包。不在while里面直接向外发也一样丢。波特率改成9600没事。
void setup() {
  Serial.begin(115200);
}
void loop() {
  while (Serial.available())   //时刻读取硬件串口数据
  {

    Serial.write(Serial.read());
    //delayMicroseconds(80); // 等待 80 us，拖时间，读完整的buffer
  }
}

### 如果正常，应该是什么样 / Expected behavior

正常情况，应该向单片机发送什么数据，就返回什么数据。

### 截图 / Screenshots

[image]

### 日志 / Logs

....

### 系统 / System

Win10 22H2

### PACK包版本 / Version

AIR MCU 0.4.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

chenxuuu, 2023-08-17:
[image]

经测试无法复现

PS：高频率的波特率，推荐使用外置时钟源，内置RC震荡没那么准

MaShuilong, 2023-08-17:
> [image]
> 
> 经测试无法复现
> 
> PS：高频率的波特率，推荐使用外置时钟源，内置RC震荡没那么准

您是否使用了外部时钟呢？

chenxuuu, 2023-08-17:
我使用的是内部时钟，所以会有概率数据错乱，但没有出现你上面写的丢包问题

版本就是当前的最新版本

HalfSweet, 2023-08-17:
应当提高主频，默认的频率是4M，无法支撑115200的高波特率中断

closed 2023-08-19 by HalfSweet

MaShuilong, 2023-08-30:
> 应当提高主频，默认的频率是4M，无法支撑115200的高波特率中断

感谢您的回答，根据您的描述提高主频后，解决问题！

profacetom, 2023-12-13:
怎么设置提高主频？