Air-duino/Arduino-AirMCU #53: 有时下载后手动复位ADC输出0

createskyblue, 2023-08-18:
### 描述一下这个bug / Describe the bug

下载后复位，analogRead的值为0
陷入该异常状态后即使硬件复位，或者重新下载程序，都无法恢复
除非断电重启

### 复现步骤 / To Reproduce

```
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println(analogRead(PA0));
}

```
下载程序后重复下面的步骤，直到异常触发
先按BOOT按键，然后按RST按键，松开RST，松开BOOT

异常触发后必须断电才能恢复正常

### 如果正常，应该是什么样 / Expected behavior

模拟值正常输出

### 截图 / Screenshots

[image]
[image]
[image]

### 日志 / Logs

-

### 系统 / System

Win10 22H2

### PACK包版本 / Version

0.4.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

HalfSweet, 2023-08-19:
可能是你读取的频率太高了，超过了ADC的最小转换时间，具体的可能是库里面存在一些BUG。

建议在两次读取之间插入一个延时，例如

```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println(analogRead(PA0));
  delay(1);
}

```

HalfSweet, 2023-08-19:
问题应该定位到了，是库的问题

closed 2023-08-20 by HalfSweet