Air-duino/Arduino-AirMCU #64: 示例 数码管 TM1637 程序中数码管只能控制一次

MethodJiao, 2023-09-02:
### 描述一下这个bug / Describe the bug

我使用 [](https://arduino.luatos.com/tutorial-extras/TM1637.html )这个中的程序，添加了一个计数器变量，但是他不会生效，貌似    tm1637.display(0, i);只有上电的第一次有效。怀疑过是内置时钟问题，尝试过各种频率，仍无法解决。

### 复现步骤 / To Reproduce

`#include "TM1637.h"
#define CLK PA_14
#define DIO PA_13
int i = 0;
TM1637 tm1637(CLK, DIO);
void setup()
{
    tm1637.init();
    tm1637.point(1);
    tm1637.set(2);
}

void loop() 
{
    tm1637.display(0, i);
    tm1637.display(1, 2);
    tm1637.display(2, 3);
    tm1637.display(3, 4);
    i++;
    delay(200);
}
`

### 如果正常，应该是什么样 / Expected behavior

第一位数字应该从0开始累加，而不是永远为0

### 截图 / Screenshots

[image]

### 日志 / Logs

无

### 系统 / System

Windows 11 企业版 22H2

### PACK包版本 / Version

0.4.5

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

HalfSweet, 2023-09-02:
如果可以的话，您是否可以提供相应的逻辑分析仪的结果呢。

这个库是使用的软件模拟，理论上不会出现时序错误的情况

MethodJiao, 2023-09-04:
> 如果可以的话，您是否可以提供相应的逻辑分析仪的结果呢。
> 
> 这个库是使用的软件模拟，理论上不会出现时序错误的情况

经尝试，当 #define CLK PA_14   #define DIO PA_13 这两个引脚时出现问题，我换成pa0和pa1问题得到解决

MethodJiao, 2023-09-04:
> 

原因不明

systemyf, 2023-10-19:
因为你程序写错了……i溢出之后就是0了。

closed 2023-11-08 by HalfSweet