Air-duino/Arduino-AirMCU #66: 能否支持FastLED？

arilink, 2023-09-05:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

FastLED是一个arduino上非常常用的库，用来控制RGB灯带或者矩阵，air001很适合做灯带控制器 希望官网可以支持一下

### 描述您想要的解决方案 / Describe the solution you'd like

FastLED支持air001

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

none

### 其他备注信息或截图 / Add any other context or screenshots about the feature request here

none

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

cqdadalele, 2023-09-15:
好像用不上去哈。

chenxuuu, 2023-09-18:
提问错地方了，应该去这里问：
https://github.com/FastLED/FastLED/issues

closed 2023-09-18 by chenxuuu