Air-duino/Arduino-AirMCU #73: Air001启用第二个硬件串口导致程序卡死或异常

Walkline80, 2023-09-25:
### 描述一下这个bug / Describe the bug

我需要使用第二个串口去连接其它设备进行通讯，使用api函数`HardwareSerial`初始化后导致`setup()`函数中的`Serial.println("setup");`无法输出

### 复现步骤 / To Reproduce

时钟源和频率选择的`24M/24M`，`TXD2`选择了`PF1`引脚，`RXD2`选择了`PF0`引脚

```cpp
#define TXD2 PF1
#define RXD2 PF0

HardwareSerial mySerial2(RXD2, TXD2);
// SoftwareSerial mySerial2(RXD2, TXD2);

void setup(void) {
    Serial.begin(19200);
    while (!Serial) {
        ;
    }

    Serial.println("setup");

    mySerial2.begin(19200);
    while (!mySerial2) {
        ;
    }
}
```

### 如果正常，应该是什么样 / Expected behavior

正常情况下应该在串口监视器打印`setup`字符串

### 截图 / Screenshots

[image]

### 日志 / Logs

串口监视器没有输出日志，所以只能贴上输出窗口的日志了

```doc
 项目使用 16560 字节（50%）的程序存储空间。最大值为 32768 字节。
 个全局变量使用 1872 个字节（45%）的动态内存，剩下 2224 个字节用于局部变量。最大值为 4096 字节。
AirISP v1.2.7.0
串口 COM13
连接中...
Chip PID is: 0x04 0x40 
擦除flash中（请耐心等待）...
擦除成功，耗时 75.6761 ms.
start write data ...
Write 16896 bytes at 0x08000000 in 14407.2715 ms

Leaving...
通过RTS硬件复位...
【提示】
若遇到因为AirMCU库或工具造成的BUG，请务必上报到下面的网址，以供开发者知晓该问题，并持续跟踪：
https://github.com/Air-duino/Arduino-AirMCU/issues
```

### 系统 / System

Win11 22H2

### PACK包版本 / Version

AirISP v1.2.7.0, AirMCU v0.4.6

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

lineleb, 2023-10-01:
是不是RX TX定义反了，我看手册里PF1应该复用成RX

Walkline80, 2023-10-01:
> 是不是RX TX定义反了，我看手册里PF1应该复用成RX

这确实是个问题，又看了下`1.0.4`版的[数据手册](https://cdn.openluat-luatcommunity.openluat.com/attachment/20230721175506039_AIR001%E8%8A%AF%E7%89%87%E6%95%B0%E6%8D%AE%E6%89%8B%E5%86%8C1.0.4.pdf)，貌似`PF1`引脚既可以作为`USART2_TX`还可以作为`USART2_RX`，`PF0`引脚也是同样的，当时选择引脚的时候应该是按照复用顺序选了最靠前的

[image]

Walkline80, 2023-10-01:
@lineleb 尝试交换了一下两个引脚定义重新烧录并运行，结果`HardwareSerial`还是不能使用，甚至之前可以凑合使用的`SoftwareSerial`也不能用了。。。。

HalfSweet, 2023-10-03:
见https://github.com/Air-duino/Arduino-AirMCU/issues/75#issuecomment-1745062071，PF0和PF1同样是同时拥有USART1和USART2的复用，导致初始化出现问题，建议强制指定使用Serial2

Walkline80, 2023-10-04:
实践证明，可以启用`Serial2`，但是`Serial2`会顶替`Serial1`的作用，结果就是`Serial1`不能使用了

Walkline80, 2023-10-04:
@HalfSweet 我贴个最简代码吧

```cpp
HardwareSerial Serial2(PF0, PF1);

void setup() {
    Serial.begin(19200);
    while (!Serial) { ; }

    Serial2.begin(19200);
    while (!Serial2) { ; }
}

void loop() {
    Serial.println("serial");
    Serial2.println("serial2");
}
```

因为我的电路已经固定了`PF0`和`PF1`，我也没有咱们的开发板，所以只能选这两个引脚了

以上代码，包括尝试增加`build_opt.h`，在`loop()`函数中，只有`Serial2.println()`可以在串口监视器输出内容，而且如果不把前边一句注释掉，大概也只能打印7~8次就停止了，但是奇怪的是串口监视器不是应该显示`Serial.println()`的内容吗？

还是说`Serial2`只能使用[默认的](https://github.com/Air-duino/Arduino-AirMCU/blob/4d4c5cf2f4f636ee09dc0806e5050a68e4709cf5/variants/AIR001/AIR001_DEV/PeripheralPins.c#L125)`PA1`和`PA0`引脚，其它不管如何设置都是使用了`Serial1`？

Walkline80, 2023-10-07:
经过反复测试，用[第二种方法](https://arduino.luatos.com/library/uart.html#airmcu-%E4%B8%93%E7%94%A8api)一样不能解决任何问题，除非指定`PA1`和`PA0`作为`Serial2`的引脚，否则无论如何设置都会映射到`Serial1`

systemyf, 2023-10-19:
实测，同样的问题发生。Serial和Serial完全会被重定义到Serial上

Walkline80, 2023-10-19:
> 实测，同样的问题发生。Serial和Serial完全会被重定义到Serial上

倒是可以用文档里写的方法组合一下解决这个问题，但还是希望官方可以解决一下

hebbelerfrank, 2023-10-21:
原来先辈们也遇到这个问题，本想用第二个串口，哎，没法用哈。

systemyf, 2023-10-21:
> 原来先辈们也遇到这个问题，本想用第二个串口，哎，没法用哈。

已经解决了，看AIRMCU的手册，它这个不同于市面上的原生arduino。

HalfSweet, 2023-10-22:
> 经过反复测试，用[第二种方法](https://arduino.luatos.com/library/uart.html#airmcu-%E4%B8%93%E7%94%A8api)一样不能解决任何问题，除非指定`PA1`和`PA0`作为`Serial2`的引脚，否则无论如何设置都会映射到`Serial1`

未成功复现，能否提供一个完整的工程？

xky183, 2023-11-01:
遇上同样的问题了,求指导
```
HardwareSerial Serial2(PB2, PA7);
void setup() {
  pinMode(PB1, OUTPUT);
  Serial.begin(9600);  // 串口输出
  Serial2.begin(9600);
}
void loop() {

  Serial.println("2111YYDS");
  digitalWrite(PIN_LED, HIGH);
  delay(1000);
  digitalWrite(PIN_LED, LOW);
  Serial2.println("Hello World!");
  while (Serial2.available() > 0)
  {
  String serialData = Serial2.readString();    // 将接收到的信息使用readString()存储于serialData变量
  Serial.println(serialData);
  }

}
```

systemyf, 2023-11-01:
please send your wechat number to my email

kevin
***@***.***

systemyf, 2023-11-01:
> 遇上同样的问题了,求指导
>
> ```
> HardwareSerial Serial2(PB2, PA7);
> void setup() {
>   pinMode(PB1, OUTPUT);
>   Serial.begin(9600);  // 串口输出
>   Serial2.begin(9600);
> }
> void loop() {
>
>   Serial.println("2111YYDS"); [...]

please send your wechat number to  systemyuan@gmail.com

xky183, 2023-11-06:
感谢systemyf指导，总结一下：
1.UART2可以用，但是只能用在PA0 PA1上，
2.在AIR001 arduino API介绍网页上例程是按照PA0PA1写的，但是没有强调只能将UART2映射到PA0PA1上，
3.我上面有问题的测试程序，改成HardwareSerial Serial2(PA1, PA0)后正常

systemyf, 2023-11-06:
> 感谢systemyf指导，总结一下： 1.UART2是可以用，但是只能用在PA0 PA1上， 2.在AIR001 arduino API介绍网页上例程是按照PA0PA1写的，但是没有强调只能将UART2映射到PA0PA1上， 3.我上面有问题的测试程序，改成HardwareSerial Serial2(PA1, PA0)后正常

是这样子，虽然也可以重载到其它引脚，但是没有这个必要。最好按照手册上默认的来连接外设，自定义反而会给项目本身管理带来麻烦。
而且请注意，Air001如果不外挂晶振，自身的Usart速率建议设置为9600，过高的波特率可能会导致丢数据。

hehe501, 2023-11-06:
是因为他官方的Serial.begin()一直都是使用的AF1复用，导致无论时Serial.begin()还是Serial1.begin()都是复用的同一个引脚，导致两个串口实际上只有一个被初始化使用

Walkline80, 2023-11-07:
> 是这样子，虽然也可以重载到其它引脚，但是没有这个必要。最好按照手册上默认的来连接外设，自定义反而会给项目本身管理带来麻烦。 而且请注意，Air001如果不外挂晶振，自身的Usart速率建议设置为9600，过高的波特率可能会导致丢数据。

我使用重载引脚的方式启用了串口2（懒得改硬件了），无外置晶振，目前设置`250000`波特率似乎并没有明显的问题，还有待观察

另外根据@HalfSweet 的描述，他并未遇到只能指定PA1和PA0作为Serial2的引脚的问题，不知道是否是官方有所保留？

systemyf, 2023-11-07:
> > 是这样子，虽然也可以重载到其它引脚，但是没有这个必要。最好按照手册上默认的来连接外设，自定义反而会给项目本身管理带来麻烦。 而且请注意，Air001如果不外挂晶振，自身的Usart速率建议设置为9600，过高的波特率可能会导致丢数据。
> 
> 我使用重载引脚的方式启用了串口2（懒得改硬件了），无外置晶振，目前设置`250000`波特率似乎并没有明显的问题，还有待观察
> 
> 另外根据@HalfSweet 的描述，他并未遇到只能指定PA1和PA0作为Serial2的引脚的问题，不知道是否是官方有所保留？

反正我自己的设计就是把官方开发板阉割了，引脚什么都按照官方默认来。一方面方便开发板上的程序直接烧到产品上，另一方面是人多了管理容易。如果你经过测试，256000不会丢数据，那就可以。这个串口波特率建议自己测下，毕竟和跳线连接方式、质量、长度都有关系。从原理上讲，最好不要超过38400。

hehe501, 2023-11-07:
我使用arduino的串口Serial.begin()情况下，使用Serial1.print（）是可以在串口中得到数据的，因此我怀疑是端口复用的问题，因为我使用keil5开发air001时，用HAL库的方式测试过，选择不对的复用方式就会导致两个串口互通

HalfSweet, 2023-11-08:
bug已复现并将在下个版本中进行修复。

Walkline80, 2023-11-08:
@HalfSweet 非常感谢你的修复，期待下个版本的更新

另外还有一个硬件串口相关的问题，在 [cores/AirMCU/HardwareSerial.h](https://github.com/Air-duino/Arduino-AirMCU/blob/a12229cfca686ab5779a323472dfed4d7c049225/cores/AirMCU/HardwareSerial.h#L45) 文件中有关于 `BUFFER_SIZE` 相关的宏定义，我理解的是我可以在我的 `.ino` 文件中事先定义并改变它们的默认值，不过在尝试编译后会有类似重复定义的错误提示（AirMCU 0.4.6），不知道有没有其它技巧可以不用去修改源文件就可以修改缓冲区默认大小？谢谢！~

HalfSweet, 2023-11-08:
当然可以，我们一般使用`build_opt`注入编译选项来改变。
在工程的同级目录下新建一个`build_opt.h`文件，然后在里面加入`-DSERIAL_RX_BUFFER_SIZE=128 -DSERIAL_TX_BUFFER_SIZE=128 `即可把缓冲区大小修改为128字节。
[image]

Walkline80, 2023-11-08:
@HalfSweet 再次感谢，已经测试成功了！~

有个小插曲，`build_opt.h`文件最后一定要留一个空行或者空格，否则会和编译时的`-e`参数拼接导致编译错误

closed 2023-11-08 by Walkline80

HalfSweet, 2023-11-08:
好的，那您可以再新建一个issue来供我们查看和修复这个问题吗

> @HalfSweet 再次感谢，已经测试成功了！~
> 
> 有个小插曲，`build_opt.h`文件最后一定要留一个空行或者空格，否则会和编译时的`-e`参数拼接导致编译错误

Walkline80, 2023-11-08:
当然没问题了^_^