Air-duino/Arduino-AirMCU #76: 编译存储空间占用优化

DWtiangu, 2023-10-07:
### 描述一下这个bug / Describe the bug

使用同一段代码，在编译arduino UNO以及air001两个不同的硬件平台时候，生成的编译程序占用空间区别较大。arduino UNO大概占用2%的空间，但是air001需要占用32%，后者是前者的十倍以上

### 复现步骤 / To Reproduce

`int led_r_pin = PB_0;
bool led_r_status = true;
void setup() {
  pinMode(led_r_pin, OUTPUT);

}

void loop() {
  digitalWrite(led_r_pin, led_r_status);
  led_r_status = !led_r_status;
  delay(1000);
}
`

### 如果正常，应该是什么样 / Expected behavior

应该减少编译时的资源占用

### 截图 / Screenshots

[image]
[image]

### 日志 / Logs

使用 0.0.1 版本的 SrcWrapper 库，在列出的文件夹中：C:\Users\haoyi-pc-15\AppData\Local\Arduino15\packages\AirM2M\hardware\AirMCU\0.2.0\libraries\SrcWrapper
"C:\\Users\\haoyi-pc-15\\AppData\\Local\\Arduino15\\packages\\AirM2M\\tools\\xpack-arm-none-eabi-gcc\\12.2.1-1.2/bin/arm-none-eabi-size" -A "C:\\Users\\haoyi-pc-15\\AppData\\Local\\Temp\\arduino\\sketches\\A6ED920F8C4F7DB75E0F418F1F929964/sketch_oct7a.ino.elf"
 项目使用 10660 字节（32%）的程序存储空间。最大值为 32768 字节。
 个全局变量使用 1120 个字节（27%）的动态内存，剩下 2976 个字节用于局部变量。最大值为 4096 字节。

### 系统 / System

win10 

### PACK包版本 / Version

aruino IDE 2.1.0     AIR MCU 0.2.0     Arduino AVR Boards 1.8.6

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

chenxuuu, 2023-10-07:
8位机与32位机本身就有区别，正常现象

DWtiangu, 2023-10-07:
希望还是能多剩余一些资源用来给用户开发吧，air001的空间相对较小，剩余空间大的话可以应用到更多的场景

DWtiangu, 2023-10-07:
[image]
包含一个显示屏的库一个串口一个GPIO之后，占用已经高达69%，后面在带上自己的应用程序，可能并不适合做开发

chenxuuu, 2023-10-07:
arm编译器就是如此，无法改变，32位机占用的空间就是要更大

HalfSweet, 2023-10-08:
32位的MCU占用指令相比8位机会更多，且GCC的优化可能会较差，目前没有更好的方案，顶多是你选择的优化用`-Os`，以及使用LTO也可以一定程度上优化flash占用

closed 2023-10-08 by HalfSweet

kaidegit, 2023-10-08:
抠flash的话用ll库开发吧

HalfSweet, 2023-10-08:
另外，没有特殊情况的话请请尽可能保持AirMCU为最新水平