Air-duino/Arduino-AirMCU #78: 实现arduino的软复位和硬件看门狗

ls701110, 2023-10-29:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

在硬件看门狗没有实现的情况下可以直接调用HAL实现看门狗吗？有没有HAL调用相关的示例，直接引用air001xx_hal_iwdg.h调用编译的时候提示找不到方法

### 描述您想要的解决方案 / Describe the solution you'd like

在硬件看门狗没有实现的情况下可以直接调用HAL实现看门狗吗？有没有HAL调用相关的示例，直接引用air001xx_hal_iwdg.h调用编译的时候提示找不到方法

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

在硬件看门狗没有实现的情况下可以直接调用HAL实现看门狗吗？有没有HAL调用相关的示例，直接引用air001xx_hal_iwdg.h调用编译的时候提示找不到方法

### 确认信息

- [X] 检查过该需求，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 这个需求是真实存在的，并且对于本项目来说是必要的
- [X] 我已经提供了该需求的清晰、详细、完整的方案，请按该方案进行实施
- [X] 我认为我提供的信息已经足够，不需要其他人继续完善该需求的其他细节
- [X] 我同意跟进该需求的实施进程，并在适时提供合理的解释与纠正

668477, 2023-11-15:
#include <air001xx_hal_iwdg.h>

int i = 0;
IWDG_HandleTypeDef watchdog;

void setup() {
  // put your setup code here, to run once:
  HAL_StatusTypeDef ret;

  watchdog.Init.Prescaler = IWDG_PRESCALER_64;
  watchdog.Init.Reload = 512;
  watchdog.Instance = IWDG;

  Serial.begin(9600);
  Serial.print("Board Start!\r\n");

  ret = HAL_IWDG_Init(&watchdog);
  Serial.println((int)ret);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  i++;
  Serial.println(i);
}

---------------------
ino文件的同目录建立build_opt.h写入一下内容
-DHAL_IWDG_MODULE_ENABLED

xky183, 2024-06-02:
上面的代码 直接编译运行，一直不看门狗复位呢

xky183, 2024-06-12:
用起来了，基于上面代码，再增加时钟

Vegetable-SYC, 2024-11-06:
楼上大佬没讲清楚，这里补充一下，如果直接复制代码烧录是没有效果的，设置低速内部时钟LSI后（如下图）芯片会不断复位，因为此时没有喂狗，如果在loop中加入HAL_IWDG_Refresh(&watchdog);（喂狗），串口打印的值会每秒递增1，即程序正常运行，代码没看懂可以找HAL库相关教程
[image]