Air-duino/Arduino-AirMCU #86: Air001若把PB6配置为SERVO引脚，会导致烧写程序困难

StreakingJerry, 2023-11-19:
### 描述一下这个bug / Describe the bug

以下Arduino程序烧录后会导致刷写新程序异常困难，重复尝试十几次才能成功烧录一次。后来发现只要把舵机引脚PB6改为其他就可以正常刷写，说明程序对引脚的引用干扰了reset后boot模式的判断。如果无法修复的话建议直接禁用PB6作为输出功能使用，若使用则触发报错

### 复现步骤 / To Reproduce

···
#include <Servo.h>

Servo servo_x;
Servo servo_y;

int pos_x = 0;
int pos_y = 0;
int ms_x = 1500;
int ms_y = 1500;
int step = 15;
int deadband = 5;

int i;
int xy[4];
int width = 168;
int height = 224;

HardwareSerial Serial2(PA1, PA0);  // RX, TX
void setup() {
  servo_x.attach(PB6);
  servo_y.attach(PB2);
  Serial1.begin(9600);
  Serial2.begin(9600);
  Serial1.println("All Done");
}

void loop() {
  if (Serial2.available() > 5) {
    if (Serial2.read() == 254) {
      for (i = 0; i < 4; i++) {
        xy[i] = Serial2.read();
      }
      if (Serial2.read() != 255) {
        Serial1.println("Error!");
      }
    }
    // Serial1.println(xy[0]); //x1
    // Serial1.println(xy[1]); //y1
    // Serial1.println(xy[2]); //x2
    // Serial1.println(xy[3]); //y2
    // Serial1.println();

    pos_x = (xy[0] + xy[2])/2;
    pos_y = (xy[1] + xy[3])/2;
  }
  if (pos_x - (width/2) > deadband) {
    ms_x += step;
  } else if ((width/2) - pos_x > deadband) {
    ms_x -= step;
  }
  if (ms_x > 2000) { ms_x = 2000; }
  if (ms_x < 1000) { ms_x = 1000; }
  Serial1.println(pos_x - (width/2));

  // for (pos_x = 0; pos_x <= 180; pos_x += 1) {  // goes from 0 degrees to 180 degrees
  //   // in steps of 1 degree
  //   servo_x.write(pos_x);  // tell servo to go to position in variable 'pos'
  //   delay(speed);           // waits 15ms for the servo to reach the position
  // }
  // for (pos_x = 180; pos_x >= 0; pos_x -= 1) {  // goes from 180 degrees to 0 degrees
  //   servo_x.write(pos_x);                  // tell servo to go to position in variable 'pos'
  //   delay(speed);                           // waits 15ms for the servo to reach the position
  // }
  // for (pos_y = 0; pos_y <= 180; pos_y += 1) {  // goes from 0 degrees to 180 degrees
  //   // in steps of 1 degree
  //   servo_y.write(pos_y);  // tell servo to go to position in variable 'pos'
  //   delay(speed);           // waits 15ms for the servo to reach the position
  // }
  // for (pos_y = 180; pos_y >= 0; pos_y -= 1) {  // goes from 180 degrees to 0 degrees
  //   servo_y.write(pos_y);                  // tell servo to go to position in variable 'pos'
  //   delay(speed);                           // waits 15ms for the servo to reach the position
  // }
}

···

### 如果正常，应该是什么样 / Expected behavior

正常烧写

### 截图 / Screenshots

连接开发板超时

### 日志 / Logs

无

### 系统 / System

WIN10

### PACK包版本 / Version

0.4.5

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

HalfSweet, 2023-11-21:
手动进入bootloader是否有效呢，还是手动也无法进入

668477, 2023-11-30:
我最近一个项目的代码也把boot0引脚用作pwm的功能，按住boot按键再rest，没有发现难进入bootloader状态的情况。