Air-duino/Arduino-AirMCU #94: Serial.write和Serial.print函数输出0xAA会变成0xFFFD

lable, 2023-12-12:
### 描述一下这个bug / Describe the bug

Serial.write和Serial.print函数输出0xAA会变成0xFFFD

### 复现步骤 / To Reproduce

Serial.write((unsigned char)0xAA);

(unsigned char) test[] = {0xAA};
Serial.write(( unsigned char*)test,sizeof(test));

### 如果正常，应该是什么样 / Expected behavior

AA

### 截图 / Screenshots

[image]

### 日志 / Logs

---- 已打开串行端口 /dev/tty.usbmodem1102 ----
fffd

### 系统 / System

macos14 arm

### PACK包版本 / Version

0.6.2

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2023-12-13:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

lable, 2023-12-13:
> ⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：
> 
> ⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:
> 
> * ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

机器人就是机器人，代码片段里已经说明了问题。上传了测试工程，你们跑跑看吧

chenxuuu, 2023-12-14:
参考
https://github.com/Air-duino/Arduino-AirMCU/issues/50

closed 2023-12-14 by chenxuuu

lable, 2023-12-14:
@chenxuuu 您验证这个问题了吗？怎么就直接关闭了。
又尝试用 LL 库芯片频率设置成了 48MHZ发送0xAA，PC 上仍然接收到的是0xFFFD，是不是硬件有什么处理，还是 0xAA有什么特殊用途？

[air001_ll_usart.ino.zip](https://github.com/Air-duino/Arduino-AirMCU/files/13667188/air001_ll_usart.ino.zip)
[image]

chenxuuu, 2023-12-14:
[image]

lable, 2023-12-14:
是vscode的arduino插件问题，用llcom能收到正确的数据，谢谢！