Air-duino/Arduino-AirMCU #95: air001 arduino 0.6.2  测试EEPROM失败？

profacetom, 2023-12-12:
### 描述一下这个bug / Describe the bug

我使用的是Air001 ,使用ARDUINO,发现用示例中EEPROM中的clear 或者put，write三个案例，进行操作，但使用eeprom__read读取的状态都是255？
我不清楚为什么？air001 的EEPROM示例是要外接2401的eeprom才能用起来还是模拟内部FLASH的？
我要怎么才能使用Aaruino中的EEprom.

### 复现步骤 / To Reproduce

使用示例中EEPROM中的clear 或者put，write三个案例任何一个，下载到air001中，执行几秒后，下载eeprom__read示例。然后打开串口监视器，查看读取的EEPROM.值，能发现一共128个字节，但值都是255?

### 如果正常，应该是什么样 / Expected behavior

期望是能打印我写入EEPROM的值。

### 截图 / Screenshots

[image]
[image]

### 日志 / Logs

11:21:48.165 -> 52	255
11:21:48.662 -> 53	255
11:21:49.175 -> 54	255
11:21:49.658 -> 55	255
11:21:50.169 -> 56	255
11:21:50.679 -> 57	255
[image]

### 系统 / System

win10

### PACK包版本 / Version

0.6.2

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

profacetom, 2023-12-13:
第一次发这种信息，有人会回复，和确认这问题吗？一般多久会官方应答？

chenxuuu, 2023-12-13:
烧录前会擦除芯片，你每次烧录都是还原

profacetom, 2023-12-13:
哦哦，谢谢。回复，那我知道怎么测试了，

profacetom, 2023-12-13:
/***
    eeprom_get example.
    确认测试通过，输出“T”写入值，读取正常。
***/
#include <EEPROM.h>
int eeAddress = 0; //EEPROM address to start reading from
struct MyObject {
  float field1;
  byte field2;
  char name[10];
};
bool putfig= 0;

void getTest() {
  MyObject customVar; //Variable to store custom object read from EEPROM.
  EEPROM.get(eeAddress, customVar);
  Serial.println("Read custom object from EEPROM: ");
  Serial.println(customVar.field1);
  Serial.println(customVar.field2);
  Serial.println(customVar.name);
}
void setup() {
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.print("put and get from EEPROM: ");
}

void loop() {
  /* Empty loop */
   if (Serial.available()) {
    char c = toupper(Serial.read());
    if (c == 'T'  ) {
      putfig=1;
    } 
  }

  if(putfig==1){
    //Data to store.
    Serial.print("put data EEPROM: ");
    MyObject putcustomVar = {
      3.14f,
      65,
      "Working!"
    };
    EEPROM.put(eeAddress, putcustomVar);
    putfig=0;
  }else{
    getTest(); //Run the next test.
  }
  delay(1000);
}

profacetom, 2023-12-13:
谢谢，大神提醒。

closed 2023-12-13 by profacetom