Air-duino/Arduino-AirMCU #96: air001 在使用arduino 中使用Serial接收数据不完整

profacetom, 2023-12-13:
### 描述一下这个bug / Describe the bug

我做了一个测试收到数据进行判断，然后放到需要的地方去，结果发现我发了五个+ “+++++  。结果显示只收到三个+++。
我截图了我的代码，和反馈效果？我认为这是个BUG ,源代码我在ESP32上测试是正常的。但搬到Air001上就反应异样了。

### 复现步骤 / To Reproduce

代码片段：
Serial.begin(115200); //uart1的初始化, 
uint8_t RxFlg=0;
uint8_t RxToBuffer( char *Cmdbuf, uint8_t *Userbuf)
{
    size_t len=0;
    //check for data from the Serial
    if(Serial.available())
       {
          
          len = Serial.available();
          uint8_t sbuf[len];
          Serial.readBytes(sbuf, len);
          if (memcmp(sbuf, "+++++", 5) == 0)
               {
                 ATFlg=1;
                 Serial.write("\n\rIn ATCmd\n\r");
               }
               if (ATFlg==1)
                  {
                    memset(Cmdbuf,0,64);
                    memcpy(Cmdbuf, sbuf,len);  
                    RxFlg=1;
                  }else{
                    memset(Userbuf,0,256);
                    memcpy(Userbuf,sbuf,len);
                    Serial.write( Userbuf,len);
                    RxFlg=1;
                  }
       }
     return len;      
}

### 如果正常，应该是什么样 / Expected behavior

输出  +++++  ,正常应该Serial.write("\n\rIn ATCmd\n\r"); 但目前返回Serial.write( Userbuf,len); 并且显示只有+++。
使用串口

### 截图 / Screenshots

[image]
[image]

### 日志 / Logs

接收的是数组很乱，
发送+++++ ，  接收到+++
发送AT+EXAT  ，接收到AT+AT
发送AT+TEST， 接收到AT+ST
发送AT+CONFIG，接收到ATCNFI
发送AT+CONFIG=3000,500,11,90，接收到AT+NFI3005001,9

### 系统 / System

win10

### PACK包版本 / Version

0.6.2

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

profacetom, 2023-12-13:
看到有别的信息，说4M主频不支持 115200 ,改成19200 发现数据接收好像是对了，但+++++还是无效，发6个++++++进入了判断，好像第一个字节有什么东西？或者结尾有不一样的东西？
at_fun[index].at_cmdName = "AT+EXAT",
 ptr = strstr((const char *)Cmdbuf, at_fun[index].at_cmdName);都比对不出来？

lable, 2023-12-14:
@chenxuuu #94 是不是和我这个是同样的问题，您下载我最新的上传的那个代码包复现一下，不要着急关闭issue，芯片的频率已经设成 48MHZ，可以重现。

chenxuuu, 2023-12-14:
参考https://github.com/Air-duino/Arduino-AirMCU/issues/50
该波特率可以考虑外挂晶振

lable, 2023-12-14:
@chenxuuu 9600波特率下也可以复现的，其它都能输出，就0xAA不行，是不是有什么特殊用途了？

```c
unsigned char test[] = {0x55, 0x00, 0x44, 0x00, 0x07, 0x72, 0x02, 0x00, 0x00, 0x01, 0x01, 0x01, 0x32, 0xAA};

// 都能正确输出
0x55, 0x00, 0x44, 0x00, 0x07, 0x72, 0x02, 0x00, 0x00, 0x01, 0x01, 0x01, 0x32 
```

chenxuuu, 2023-12-14:
https://github.com/Air-duino/Arduino-AirMCU/issues/50#issuecomment-1681605278

0xaa可以显示

lable, 2023-12-14:
[image]

> @chenxuuu 9600波特率下也可以复现的，其它都能输出，就0xAA不行，是不是有什么特殊用途了？
> 
> ```c
> unsigned char test[] = {0x55, 0x00, 0x44, 0x00, 0x07, 0x72, 0x02, 0x00, 0x00, 0x01, 0x01, 0x01, 0x32, 0xAA};
> 
> // 都能正确输出
> 0x55, 0x00, 0x44, 0x00, 0x07, 0x72, 0x02, 0x00, 0x00, 0x01, 0x01, 0x01, 0x32 
> ```

您用我这个代码跑下看看，下载压缩包里。
[air001_ll_usart.ino.zip](https://github.com/Air-duino/Arduino-AirMCU/files/13667300/air001_ll_usart.ino.zip)

lable, 2023-12-14:
@chenxuuu NND 居然是vscode的arduino插件问题，用llcom能收到正确的数据，不好意思！

profacetom, 2023-12-20:
感谢，我尝试调整一下使用的主频，我是用的默认4M主频的，换成24M的试试，是不是就能正常了

profacetom, 2024-01-02:
还是不行，是不是我使用DAPLink Air32F103CBT6的 9.9元买的。导致发送的数据是不对的。怎么在+++++后面有个方块？
[image]

chenxuuu, 2024-01-02:
[image]
用sscom还能看不懂结尾的方框

profacetom, 2024-01-03:
哈哈，是回车换行啊？但不对啊，奇怪了，用ESP32就没这些东西的，代码也正常比对，但AIR001就不执行的？不理解，我也是个三脚猫，画瓢可以，找问题不在行啊！

chenxuuu, 2024-01-03:
需要看原始的二进制数据

closed 2024-01-04 by chenxuuu