Air-duino/Arduino-AirMCU #97: 328P与air001编译相同项目，flash空间大一倍之多，请求优化。

caoxuan19961206, 2023-12-14:
### 描述一下这个bug / Describe the bug

我尝试将一个应用在328p上的项目移植到AIR001上，但是328P的FLASH空间占用47%。但使用AIR001，FLASH空间超出33%.

### 复现步骤 / To Reproduce

已使用u8glib库例程，都可复现。

### 如果正常，应该是什么样 / Expected behavior

希望能够更新pack包，解决此问题，以便此MCU应用于更多场景。

### 截图 / Screenshots

希望能够更新pack包，解决此问题，以便此MCU应用于更多场景。

### 日志 / Logs

希望能够更新pack包，解决此问题，以便此MCU应用于更多场景。

### 系统 / System

windows7x64

### PACK包版本 / Version

0.6.2

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

chenxuuu, 2023-12-15:
32位arm占用就是这么多，编译器决定

LuaRobot, 2023-12-15:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

HalfSweet, 2023-12-15:
可以考虑开启`-lto`优化，32bit的mcu指令多一些是正常的

closed 2023-12-15 by HalfSweet