Air-duino/Arduino-AirMCU #98: Air32F103在arduino怎么下载

cqdadalele, 2023-12-31:
### 描述一下这个bug / Describe the bug

不是bug,主要是只有两个选项bug和新建议。所以只有写BUG了？

### 复现步骤 / To Reproduce

001下载是没问题的，但是AIR32F103如何才能下载进去呢？用TTL 串口连接到A9 A10，手动按boot0和RESET，没有效果。
用DPLINK在KEIL中也能正常下载，但是在arduino中也无法下载。

### 如果正常，应该是什么样 / Expected behavior

正常下载

### 截图 / Screenshots

[image]

### 日志 / Logs

无

### 系统 / System

win10

### PACK包版本 / Version

1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2023-12-31:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction
- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

HalfSweet, 2023-12-31:
可以尝试降低波特率，检查TX和RX有没有接反，保证器件此时处于bootloader模式中