Air-duino/Arduino-AirMCU #100: 0.6.3版本安装报错

WellsWang, 2024-01-01:
### 描述一下这个bug / Describe the bug

arduino IDE Board Manager中安装Air MCU 0.6.3版本报错
Failed to install platform: 'Air MCU:0.6.3'. 13 INTERNAL: Cannot install platform: installing platform AirM2M:AirMCU@0.6.3: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 6913613 != 6913615

### 复现步骤 / To Reproduce

Windows 11环境，在Board Manager中安装，Arduino IDE 2.2.1

### 如果正常，应该是什么样 / Expected behavior

被正常安装

### 截图 / Screenshots

[image]

### 日志 / Logs

Failed to install platform: 'Air MCU:0.6.3'. 13 INTERNAL: Cannot install platform: installing platform AirM2M:AirMCU@0.6.3: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 6913613 != 6913615

### 系统 / System

Windows 11

### PACK包版本 / Version

0.6.3

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-01-01:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

HalfSweet, 2024-01-02:
删掉重试即可，应该是cdn的缓存问题，或者可以尝试用这个地址 https://github.com/Air-duino/Arduino-pack-json-ci/releases/download/Nightly/package_air_index.json

closed 2024-01-04 by HalfSweet