Air-duino/Arduino-AirMCU #102: Air001 arduino ide windows上传程序出现报错 苹果上传没有问题

aimerl, 2024-01-08:
### 描述一下这个bug / Describe the bug

我在windows多台电脑都上传不了程序 出现报错 
但在苹果电脑就可以上传程序

### 复现步骤 / To Reproduce

首先点上传直接出现上传失败

### 如果正常，应该是什么样 / Expected behavior

上传成功

### 截图 / Screenshots

[image]

### 日志 / Logs

上传失败: 上传错误：exit status 0xc0000135

### 系统 / System

Windows 10 专业版

### PACK包版本 / Version

0.63

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-01-08:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

chenxuuu, 2024-01-08:
把输出框里的日志全部发出来，日志不全

chenxuuu, 2024-01-08:
手动安装vc++2015运行库