Air-duino/Arduino-AirMCU #104: 开启LTO后，401固件编译出错

lllwan, 2024-01-26:
### 描述一下这个bug / Describe the bug

SrcWrapper/air/analog.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/bootloader.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/clock.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/core_callback.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/dwt.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/hw_config.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/interrupt.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/otp.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/pinmap.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/system_airyyxx.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/timer.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/uart.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/new.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/syscalls.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/PeripheralPins.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/generic_clock.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/qfp_fix.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/qfplib-m0-full.S.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/variant_generic.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/cores/f3ce37e850b964bd1ddbd76c2790882d/core.a -lc -Wl,--end-group -lm -lgcc -lstdc++
/Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-fstatr.o): in function `_fstat_r':
(.text._fstat_r+0xe): undefined reference to `_fstat'
/Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-isattyr.o): in function `_isatty_r':
(.text._isatty_r+0xc): undefined reference to `_isatty'
collect2: error: ld returned 1 exit status

使用 0.0.1 版本的 SrcWrapper 库，在列出的文件夹中：/Users/my/Library/Arduino15/packages/AirM2M/hardware/AirMCU/0.6.3/libraries/SrcWrapper
exit status 1

Compilation error: exit status 1

### 复现步骤 / To Reproduce

开启LTO后点击编译即可复现

### 如果正常，应该是什么样 / Expected behavior

开启LTO后，401固件编译出错

### 截图 / Screenshots

[image]

### 日志 / Logs

SrcWrapper/air/analog.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/bootloader.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/clock.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/core_callback.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/dwt.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/hw_config.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/interrupt.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/otp.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/pinmap.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/system_airyyxx.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/timer.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/air/uart.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/new.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/libraries/SrcWrapper/syscalls.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/PeripheralPins.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/generic_clock.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/qfp_fix.c.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/qfplib-m0-full.S.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/sketches/0A5C4ED4A17A2378B6FDCEA83D174E4A/core/variant_generic.cpp.o /private/var/folders/7x/25_3v9cj6tv4h2w6c49crj4w0000gn/T/arduino/cores/f3ce37e850b964bd1ddbd76c2790882d/core.a -lc -Wl,--end-group -lm -lgcc -lstdc++
/Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-fstatr.o): in function `_fstat_r':
(.text._fstat_r+0xe): undefined reference to `_fstat'
/Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: /Users/my/Library/Arduino15/packages/AirM2M/tools/xpack-arm-none-eabi-gcc/13.2.1-1.1/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc_nano.a(libc_a-isattyr.o): in function `_isatty_r':
(.text._isatty_r+0xc): undefined reference to `_isatty'
collect2: error: ld returned 1 exit status

使用 0.0.1 版本的 SrcWrapper 库，在列出的文件夹中：/Users/my/Library/Arduino15/packages/AirM2M/hardware/AirMCU/0.6.3/libraries/SrcWrapper
exit status 1

Compilation error: exit status 1

### 系统 / System

MACOS

### PACK包版本 / Version

1.0.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-01-26:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

chenxuuu, 2024-01-26:
air401都没发布，包没做完呢

closed 2024-01-26 by chenxuuu