Air-duino/Arduino-AirMCU #106: air001 ADC抖动

xitonghan, 2024-03-25:
### 描述一下这个bug / Describe the bug

我用他的ADC 接口， 但是抖动特别厉害 
除了用数学平均法还有什么改善的办法？

void setup() {
  Serial.begin(115200);
  analogReadResolution(12);
}

void loop() {
  Serial.printf("PA0 Voltage is %d mv \n", analogReadMillivolts(PA0));
  Serial.printf("PA1 Voltage is %d mv \n", analogReadMillivolts(PA1));
  Serial.printf("Chip temp is %d C \n", analogReadTempSensor());
  Serial.printf("Vref is %d mv \n", analogReadVref());
  delay(500);
}

### 复现步骤 / To Reproduce

获得ADC数据，传输到 串口，或者液晶屏 
VREF temp  IO 电压 都是一样的抖

### 如果正常，应该是什么样 / Expected behavior

我用的 单片机例如ESP8266  uno  没有如此刺激的抖动！

### 截图 / Screenshots

[image]

### 日志 / Logs

看上图吧 

### 系统 / System

Arduino AirMCU

### PACK包版本 / Version

1.01

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-03-25:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

xitonghan, 2024-03-25:
[image]
补充日志图片

chenxuuu, 2024-04-26:
没什么改善方法，精度就是不高

closed 2024-04-26 by chenxuuu

Hello-2877515114, 2024-09-09:
我这个问题，固件刚下载然后复位，ADC非常的准，掉电之后就不行了