Air-duino/Arduino-AirMCU #107: PB6设置为上拉输入状态时异常

sunnyguhz, 2024-04-16:
### 描述一下这个bug / Describe the bug

项目中需要用到PB6的上拉输入状态，但是程序一直无法正常运行。后来发现是PB6的上拉输入状态有问题。
单独新建一个Arduino项目，3.3V供电，设置PB6为上拉输入状态，PB6在没有任何外部下拉的情况下用万用表测电压只有1.6V，处于不上不下的状态。
试过其他引脚都能在上拉输入状态维持接近3.3V的状态，包括设置PF4也能正常接近3.3V。只有PB6有问题。
已使用多只万用表测量，结果一致，排除万用表的问题。确认就是PB6无法上拉输入。

### 复现步骤 / To Reproduce

使用以下代码可复现PB6无法上拉输入的情况
```
void setup() {
  pinMode(PB6, INPUT_PULLUP);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### 如果正常，应该是什么样 / Expected behavior

正常情况当然应该是PB6接近3.3V

### 截图 / Screenshots

没有截图

### 日志 / Logs

没有日志

### 系统 / System

WIN10  22H2

### PACK包版本 / Version

0.6.4

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-04-16:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

xky183, 2024-06-03:
无法上拉的原因也许是外部下拉阻值小于内部上拉，所以总体表现出来你的上拉失效