Air-duino/Arduino-AirMCU #108: debian12系统arduino-ide_2.3.2无法安装air mcu开发板

be-engineer, 2024-04-23:
### 描述一下这个bug / Describe the bug

在debian系统下下载了arduino-ide_2.3.2_Linux_64bit.AppImage，运行后安装air001开发板，结果报错

### 复现步骤 / To Reproduce

arduino-ide_2.3.2_Linux_64bit.AppImage，打开左边的“开发板管理器”
[image]
点击安装

### 如果正常，应该是什么样 / Expected behavior

正常应该能安装air mcu的开发板

### 截图 / Screenshots

[image]

### 日志 / Logs

```
Downloading packages
AirM2M:xpack-arm-none-eabi-gcc@13.2.1-1.1
AirM2M:CMSIS@5.7.0
AirM2M:AirISP@0.2.0
Failed to install platform: 'AirM2M:AirMCU:0.6.4'.
Error: 2 UNKNOWN: Get "https://arduino.luatos.com/AirISP-0.2.0-linux-x86_64.tar.gz": tls: failed to verify certificate: x509: certificate signed by unknown authority
```

### 系统 / System

debian12

### PACK包版本 / Version

0.6.4

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-04-23:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

chenxuuu, 2024-04-23:
更新你的ca-certificates

 

 
 
 
 image.png (view on web)
 
 
 
 
image.png (view on web)
 
 Downloading packages ***@***.*** ***@***.*** ***@***.*** Failed to install platform: 'AirM2M:AirMCU:0.6.4'. Error: 2 UNKNOWN: Get "https://arduino.luatos.com/AirISP-0.2.0-linux-x86_64.tar.gz": tls: failed to verify certificate: x509: certificate signed by unknown authority

closed 2024-04-26 by chenxuuu