Air-duino/Arduino-AirMCU #114: 安装开发板失败 Downloading packages AirM2M:xpack-arm-none-eabi-gcc@13.2.1-1.1 AirM2M:CMSIS@5.7.0 AirM2M:AirISP@0.2.0 Failed to install platform: 'AirM2M:AirMCU:0.6.4'. Error: 2 UNKNOWN:

keyboard2005, 2024-09-13:
### 描述一下这个bug / Describe the bug

Downloading packages
AirM2M:xpack-arm-none-eabi-gcc@13.2.1-1.1
AirM2M:CMSIS@5.7.0
AirM2M:AirISP@0.2.0
Failed to install platform: 'AirM2M:AirMCU:0.6.4'.
Error: 2 UNKNOWN: 

### 复现步骤 / To Reproduce

...

### 如果正常，应该是什么样 / Expected behavior

...

### 截图 / Screenshots

[image]

### 日志 / Logs

Downloading packages
AirM2M:xpack-arm-none-eabi-gcc@13.2.1-1.1
AirM2M:CMSIS@5.7.0
AirM2M:AirISP@0.2.0
Failed to install platform: 'AirM2M:AirMCU:0.6.4'.
Error: 2 UNKNOWN: 

### 系统 / System

win10 20h2

### PACK包版本 / Version

0.6.4

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-09-13:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供复现步骤信息 does not provide steps for reproduction
- ❌未提供复现所需的代码或工程 does not provide codes or project for reproduction

MicroD, 2024-09-29:
撤了吧，这家公司快垮了或者已经垮了，没人维护的，开发包都装不上了

closed 2024-10-01 by keyboard2005