Air-duino/Arduino-AirMCU #115: 使用软件串口编译上传之后程序不运行

Aivin40, 2024-10-09:
### 描述一下这个bug / Describe the bug

使用软件串口编译上传之后程序不运行<SoftwareSerial.h>这个库

### 复现步骤 / To Reproduce

正常的使用软串口用一个ttf转串口模块连接到开发板上运行<SoftwareSerial.h>的程序,经过测试发现程序中如果任何带有软串口的代码都会停止运行,正常的载入软串口的库是没有问题的

### 如果正常，应该是什么样 / Expected behavior

no

### 截图 / Screenshots

no

### 日志 / Logs

不打印任何信息

### 系统 / System

win10

### PACK包版本 / Version

1.0.1.1

### 验证

- [X] 检查过该问题，之前没有人提过 / Check that there isn't already an issue that reports the same bug to avoid creating a duplicate.
- [X] 提供了最小可复现工程或详细的复现步骤，确保开发者可以复现 / The provided reproduction is a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example) of the bug.
- [X] 已经提供了完整的报错信息、日志、截图，没有经过删减。

LuaRobot, 2024-10-09:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供出错时需要的日志文件 does not provide logs when bug happening

chenxuuu, 2024-10-10:
经测试无问题，可以提供可复现的代码后重新开启本issue

closed 2024-10-10 by chenxuuu